betaflight/betaflight #14232: MacOS Build is Broken

ahalekelly, 2025-02-02:
### Describe the bug

I'm trying to build from source on MacOS and I get the following error:

```
$ make MATEKH743
Building target config MATEKH743
/Library/Developer/CommandLineTools/usr/bin/make -j hex CONFIG=MATEKH743
mk/config.mk:37:  unterminated call to function shell': missing )'.  Stop.
make:  [MATEKH743] Error 2

```

mk/config.mk:37 is the following

`HSE_VALUE_MHZ := $(shell sed -E -n "/^\s#\sdefine\s+SYSTEM_HSE_MHZ\s+([0-9]+).*/s//\1/p" $(CONFIG_HEADER_FILE))`

MATEKH743 has no SYSTEM_HSE_MHZ value defined but AIKONF4 does.

Testing this sed command returns nothing

`sed -E -n "/^\s#\sdefine\s+SYSTEM_HSE_MHZ\s+([0-9]+).*/s//\1/p" src/config/configs/AIKONF4/config.h`

Claude thinks this is because the sed syntax used is for GNU sed and isn't compatible with the BSD sed used on MacOS and suggested trying

`sed -n 's/^[[:space:]]#[[:space:]]define[[:space:]]SYSTEM_HSE_MHZ[[:space:]]([0-9][0-9])./\1/p' src/config/configs/AIKONF4/config.h`

Which returns 8 as defined in the config file, seems like progress. But when I update line 37 in config.mk with this, I still get the same error

`HSE_VALUE_MHZ := $(shell sed -n 's/^[[:space:]]#[[:space:]]define[[:space:]]SYSTEM_HSE_MHZ[[:space:]]([0-9][0-9])./\1/p' $(CONFIG_HEADER_FILE))`
 
I get this error on Git latest but not on v4.5.1, it looks like @kedeng added the sed command to this line a week ago here: https://github.com/betaflight/betaflight/commit/059ada4e4b9a27182d287df02fdc57327288f9aa

### To Reproduce

Follow these steps: https://betaflight.com/docs/development/building/Building-in-Mac-OSX

### Expected behavior

Should build properly

### Support ID

```txt
N/A
```

### Flight controller

N/A

kedeng, 2025-02-03:
I'll investigate.

kedeng, 2025-02-03:
The # symbol in the shell needs to be escaped.

closed 2025-02-05 by haslinghuis