betaflight/betaflight #14238: DMA header refactor

ggabriel-airmod, 2025-02-04:
### Describe the bug

in dma.h there is multiple definition of structures and macro depending the target MCU.

nevertheless #if, #else, #elif and #endif are confusing and need a small clean-up to be more readable ( mainly when we need to add a new MCU target)

### To Reproduce

try to add a new MCU target

### Expected behavior

a clean way to know for each  MCU target how the structure and macro are associated

### Support ID

```txt
DMA
```

### Flight controller

STM32

ggabriel-airmod, 2025-02-04:
dma_apm32.h contains exactly the same data as dma.h for MCU that use `DMA_Stream_TypeDef` structure and have `USE_HAL_DRIVER` defined.

the idea to extract specific dma out of dma.h is interresting, as it's split some information, 
nevertheless, instead of dma_apm32.h we could split into dma_steam_type.h, dma_channel_type.h and keep generic in dma.h

also, for each MCU family, we can define in platform_mcu.h `#define USE_DMA_STREAM_TYPE` or `#define USE_DMA_CHANNEL_TYPE` to simplify the dma.h readablility.

ledvinap, 2025-02-04:
@ggabriel-airmod : Removing duplication without creating unnecessary dependencies will definitely work. 
Just use different naming - INO `USE_` should be used only for configurable settings (USE_HAL_DRIVER is border case - theoretically there can be HAL and stdperiph driver for given target). There is already DMA_TRAIT_CHANNEL / DMA_TRAIT_MUX that is used in ./platform/common/stm32/serial_uart_hw.c. (there is still time to change `_TRAIT_` naming. I'm not native speaker, there may be better option)

There are other problems with DMA - `#define` macros were used as compile-time database to configure STM32F1 DMA at compile time.  IIRC current code does not use static DMA mapping (and there is no restriction on most platforms), but huge complicated mapping structures are still build. A lot of code san be simply discarded or replaced with much simpler mechanism.

ggabriel-airmod, 2025-02-04:
alright, `_TRAIT_` seems good for me (i'm not native too).

I will check if it's not huge risk for regression about the DMA mapping refactor.

ggabriel-airmod, 2025-02-04:
I have checked the some ST DMA structure element (like DMA_Stream_TypeDef and DMA_Channel_TypeDef), 
it's the same between F4 and F7 but they changed elements name (ie. SCFG to CR)
and they changed from Stream to Channel with G4 (and H5 so)

So, I will, in the idea of small pull request, only create those files with `_TRAIT_`.

blckmn, 2025-02-04:
This file should not be platform specific. Ie generic DMA methods only.

The platform specific dma requirements should move to /src/platform

DMA platform specific code should be abstracted from the BF code.

ggabriel-airmod, 2025-02-05:
indeed, currently dma.h is full of `#if defined(platform1) || defined(platform3)` etc.
the idea is to move platform specific stuff inside `platform`, and potentially common/similar stuff into `platform/common`

ggabriel-airmod, 2025-02-05:
there is one point i'm not truly sure about : 

`USE_ATBSP_DRIVER` is only defined for AT32 platform, nevertheless, in lot of STM32 sources, there is some `#if defined(USE_ATBSP_DRIVER)`

is it a legacy of a previous usage of stm32 for AT32 before split it into a separated platform ?

if it's the case, I can also clean this.

blckmn, 2025-02-05:
Anything in common/stm32 is shared between stm, apm and at

closed 2025-03-15 by github-actions[bot]