betaflight/betaflight #14240: AltHold unstable on 3 inch quad on BF4.6

macgyverek, 2025-02-04:
### Describe the bug

I have a problem with AltHold stability on my 3-inch drone. The throttle control is very unstable, causing the model to either drop quickly or climb rapidly. After a while, it stabilizes, but you can hear that the throttle is not being controlled smoothly. The throttle level is correctly set, and I use around 1290 for hovering. I’m including my dump and the logs from the last flight. PosHold works stably.

I have an FC with a foam-protected barometer and a Matek M8Q-5883 GPS.

I attached logs:

[btfl_007.zip](https://github.com/user-attachments/files/18662037/btfl_007.zip)

### To Reproduce

Use AltHold on 3" drone with 1404 3850KV motors

### Expected behavior

Stabile AltHold

### Support ID

```txt
Id: "42a430bf-ba3e-4ddf-8c58-b1163b1d4e3d"
```

### Flight controller

MAMBAF722_I2C

### Other components

Motors: 1404 3850KV
GPS: Matek M8Q-5883

macgyverek, 2025-02-04:
I also attached a prints screen from logs; this is a moment when I switch flight modes from acro to atlhold:
[image]

nerdCopter, 2025-02-04:
i will ping @ctzsnooze  here, but GitHub is for code-development. You should rather request support in [Discord](https://discord.betaflight.com/) and i assume this is a tuning issue of alt-hold parameters as per : https://betaflight.com/docs/wiki/guides/current/position-hold-v4-6#altitude-hold

macgyverek, 2025-02-04:
But in my opinion its something wrong with altitude reading/estimation. Look into my logs print screen.

closed 2025-03-15 by github-actions[bot]

haslinghuis, 2025-03-15:
Try lowering trust weight of using baro

```
set altitude_prefer_baro = 100
```

haslinghuis, 2025-03-15:
Related `altitude` settings

```bash
altitude_source = DEFAULT
Allowed values: DEFAULT, BARO_ONLY, GPS_ONLY

altitude_prefer_baro = 100
Allowed range: 0 - 100

altitude_lpf = 300
Allowed range: 10 - 1000

altitude_d_lpf = 100
Allowed range: 10 - 1000

ap_landing_altitude_m = 4
Allowed range: 0 - 200

ap_altitude_p = 15
Allowed range: 0 - 200

ap_altitude_i = 15
Allowed range: 0 - 200

ap_altitude_d = 15
Allowed range: 0 - 200

ap_altitude_f = 15
Allowed range: 0 - 200
```