betaflight/betaflight #14242: HGLCF411 no work UBLOX on 4.5.1, only NMEA on 9600 speed :(

gumileeev, 2025-02-05:
### Is your feature request related to a problem? Please describe

In the program version 4.3.0, my GPS works in Ublox mode and in NMEA mode. NMEA works at all speeds. When I install the program version 4.5.1, my GPS works only in NMEA mode and only at a speed of 9600. I would like to use Ublox, because only it is really optimized in 4.5.1.
I use a GPS from a Matek M8Q-5883.
In my flight controller (Zeus 5 AIO), the UART1 was used for WiFi from the factory, but I re-soldered it to use GPS.
I am attaching working configurations 4.3.0 and 4.5.1
Please tell me how I can return the Ublox mode?

### Describe the solution you'd like

I really want to restore the GPS operation in Ublox mode on the program version 4.5.1

### Describe alternatives you've considered

I don’t see any alternative, because 4.5.1 has many useful changes for GPS, and going back to 4.3.0 is not a solution.

gumileeev, 2025-02-06:
I did a little investigation, installed previous versions and uploaded a backup file of version 4.3.0 and found out that ublox stops working on version 4.4.2 and does not work any more. On version 4.4.0 ublox still works.
Why don't I try a backup from the 4.5.1 file? Because some critical changes were made to the config to switch to this version: "resource SOFTSERIAL_TX 1 B03" instead of "resource SERIAL_TX 11 B03", dshot300 instead of dshot600 and pid 4k instead of 8k. That's why I try a backup from 4.3.0 and see where ublox still works.

gumileeev, 2025-02-06:
Perhaps my connection method is not the best, I took the photo from the Internet, but the connection diagram is exactly the same. This can be condemned, but Ublox worked up to 4.4.0, but stops working on 4.4.2, so I assume that the problem is not in the connection method.

[image]

gumileeev, 2025-02-06:
The same behavior if I connect GPS Beitian BN-180. On all versions the behavior of receivers is repeated.

haslinghuis, 2025-02-06:
- Try 4.6 development version
- Try lowering `gps_update_rate_hz = 10` to 5 (experiment with other values)

gumileeev, 2025-02-06:
> * Try 4.6 development version
> * Try lowering `gps_update_rate_hz = 10` to 5 (experiment with other values)

Version 4.6 unfortunately didn't change anything.
I don't know, I first lowered the rate to 5, then tried 1, but nothing changed.
I'm ready to try experiments as much as it takes.
Id: "579099ff-94c4-4d72-aac3-7dec0c48af05"

gumileeev, 2025-02-06:
I also made a support ID from version 4.4.0, where everything works.
Id: "9d374bac-68f3-482f-b5a6-ab4c6fc439ab"

haslinghuis, 2025-02-06:
> GPS: NOT CONNECTED, UART1 38400 (set to 9600), NOT CONFIGURED, version =  unknown
Arming disable flags: RXLOSS CLI MSP

- try 38400 baudrate (might require cold boot to activate)
- try disabling auto config
- check ucenter[2]

gumileeev, 2025-02-06:
> > GPS: NOT CONNECTED, UART1 38400 (set to 9600), NOT CONFIGURED, version =  unknown
> > Arming disable flags: RXLOSS CLI MSP
> 
> * try 38400 baudrate (might require cold boot to activate)
> * try disabling auto config
> * check ucenter[2]

Changing the speed and disabling auto-tuning did not help. Then I went to the cli and made the command gpspassthrough. Then I closed the configurator and opened the y-center. I attach a screenshot.
Id: "0c0e9379-eb51-4b0f-8f29-c0b9b820c56f"

[image]

gumileeev, 2025-02-07:
I am adding the result from the center-2 program:

[image]

gumileeev, 2025-02-07:
> > GPS: NOT CONNECTED, UART1 38400 (set to 9600), NOT CONFIGURED, version =  unknown
> > Arming disable flags: RXLOSS CLI MSP
> 
> * try 38400 baudrate (might require cold boot to activate)
> * try disabling auto config
> * check ucenter[2]

I am adding the result from the center-2 program:

[image]

gumileeev, 2025-02-07:
> * Try 4.6 development version

I found the exact place where the problem appears. I studied each commit on the page https://github.com/betaflight/betaflight/compare/4.4.0...4.4.2 and installed it on a quadcopter to check.
The problem was not in the first commit from May 11, but in the second, third and fourth - the GPS no longer wants to work in Ublox mode.
I am not a programmer, and I do not know what exactly affected the work. If you can send me a corrected program file, version 4.6.0 or 4.5.1 - I could check. I will be glad if we can fix the problem together.

[image]

haslinghuis, 2025-02-07:
You didn't mention if you updated the configurator ? Or even better use https://app.betaflight.com (requires chrome based browser)

gumileeev, 2025-02-07:
> You didn't mention if you updated the configurator ? Or even better use https://app.betaflight.com (requires chrome based browser)

For working with version 4.6.0 I used https://app.betaflight.com/ . For all other versions I used the desktop version of the configurator 10.10.0

haslinghuis, 2025-02-07:
Checking BN220 GPS on 4.6:

```
GPS: connected, UART2 57600 (set to 57600), configured, version =  M8
```

mituritsyn, 2025-02-07:
> Checking BN220 GPS on 4.6:
> 
> ```
> GPS: connected, UART2 57600 (set to 57600), configured, version =  M8
> ```

Does it matter if a person found a specific commit after which his setup doesn't work, the commit clearly had uart related changes.

gumileeev, 2025-02-07:
> Checking BN220 GPS on 4.6:
> 
> ```
> GPS: connected, UART2 57600 (set to 57600), configured, version =  M8
> ```

I installed 4.6.0, I check on bn-180 (m8k behaves the same)
From the settings I entered only the line
```
resource MOTOR 5 NONE
resource SERIAL_TX 1 B03
save
```
After that I turned on GPS on the port uart 1 and turned on the speed 57600
But only Nmea works, and Ublox does not want to work.
I also checked all the commits until I found at what point Ublox stops working for me.

Id: "573b42ce-a123-4ad6-8b71-b81e55646084"

haslinghuis, 2025-02-07:
@mituritsyn 
Of course - highly appreciated, but breaking PR had some follow ups too.

@gumileeev try removing timer and dma from B03

```bash
timer B03 none
dma B03 none
save
```

gumileeev, 2025-02-07:
> 
> [@gumileeev](https://github.com/gumileeev) try removing timer and dma from B03
> 
> timer B03 none
> dma B03 none
> save

```
..Timer OK..
# dma B03 none
###ERROR IN dma: BAD DEVICE: B03###
save
```
Nothing has changed: Nmea works, Ublox does not work.
Id: "b063f268-0d5f-41b4-ba7c-7df5fda7e33a"

haslinghuis, 2025-02-07:
Excuse me, use `dma pin B03 none`

gumileeev, 2025-02-07:
> Excuse me, use `dma pin B03 none`

No, it didn't help.
Id: "6b16972c-0fc7-4a03-8c9f-3ee9643d5384"

haslinghuis, 2025-02-07:
Hmm, now it reports 115K2 instead as earlier it was 38400

```
GPS: NOT CONNECTED, UART1 115200 (set to 57600), NOT CONFIGURED, version =  unknown
```

Assume we want ublox to work.
One change at a time (cold boot in between)
- Try setting 115K200 in ports tab.
- Try disabling autoconfig

gumileeev, 2025-02-07:
> Hmm, now it reports 115K2 instead as earlier it was 38400
> 
> ```
> GPS: NOT CONNECTED, UART1 115200 (set to 57600), NOT CONFIGURED, version =  unknown
> ```
> 
> Assume we want ublox to work. One change at a time (cold boot in between)
> 
> * Try setting 115K200 in ports tab.
> * Try disabling autoconfig

It didn't work
Id: "4fde413c-e4e8-42d4-abc0-cda7d3599884"

haslinghuis, 2025-02-07:
Using debug GPS_CONNECTION I see this is sensor tab

[image]

gumileeev, 2025-02-07:
> Using debug GPS_CONNECTION I see this is sensor tab
> 
> [image]

I don't understand what this means? Should I do something?

haslinghuis, 2025-02-07:
To check same on your side:
- Enable GPS_CONNECTION debug mode in blackbox tab
- Make a screenshot of sensors tabs when the page had obtained it first data like in the picture.

gumileeev, 2025-02-07:
> To check same on your side:
> 
> * Enable GPS_CONNECTION debug mode in blackbox tab
> * Make a screenshot of sensors tabs when the page had obtained it first data like in the picture.

[image]
[image]

haslinghuis, 2025-02-07:
Please re-enable gps autoconfig - as baudrate negotiation is going all over the place and task time going through the roof.

gumileeev, 2025-02-07:
> Please re-enable gps autoconfig - as baudrate negotiation is going all over the place and task time going through the roof.
> 
> uCenter reports baudrate as 38400. Maybe we stick with that setting.

I will also add a new identifier so that you can check the correctness of my configuration.
Id: "8db49e5d-b2c8-4c2f-aa86-f00932490de5"
[image]

haslinghuis, 2025-02-07:
... and also change gps port to 38400 (we had that before but with autoconfig disabled)

gumileeev, 2025-02-07:
> ... and also change gps port to 38400 (we had that before but with autoconfig disabled)
> 
> I also notice it worked before with 57K6 ...
> 
> [@ledvinap](https://github.com/ledvinap) any idea ?

That's right, on 57600 everything works fine on other versions of the program too.
Id: "bc26c9fc-ed6b-4ae7-a553-75f4eb316652"

[image]
[image]

gumileeev, 2025-02-10:
@SteveCEvans Can I please ask you to look at the [code fix from May 11, 2023](https://github.com/betaflight/betaflight/commit/ca12a4c611408e3b2b1f8320bf9ea30abb20b156). Since then, Ublox has stopped working on my configuration. On any version before this date, it works.

ledvinap, 2025-02-10:
@gumileeev: using pullup on gps rx (tx on  BF side) will probaby help

gumileeev, 2025-02-11:
> [@gumileeev](https://github.com/gumileeev): using pullup on gps rx (tx on BF side) will probaby help

I tried to make a pull-up, for this I took a 4.7k resistor and placed it between the fc thx (gps rh) and +5v (it is in the photo nearby, power appears on it immediately after connecting the usb). There is no result. I tried to reduce the speed to 9600 and tried to disable auto-tuning, but this does not help. If I switch to NMEA - it works immediately, on UBLOX it does not work again.

gumileeev, 2025-02-13:
@haslinghuis will this be considered a bug? is there a chance that the code fix will be revised? in the vedb code, changes were clearly made to the uart, and after that, nmea only works at 9600. You did so much work to optimize ublox, and in the end, only nmea 9600 remained to work :(

gumileeev, 2025-02-26:
@haslinghuis @ledvinap @mituritsyn @SteveCEvans I have new data to help developers find a bug in the code. 
Perhaps earlier you could be confused by the fact that I connect GPS instead of WiFi, which was from the manufacturer. I hoped that the problem would disappear if I reassign the ports and use LED for TX1 and S5 for RX1, this connection method is [mentioned for my board here](https://rotorbuilds.com/build/26858). 
I hoped that if it works, then I will do without fixing the code. I re-soldered GPS to new pins, hoping that this will eliminate my problems. 
And what do you think? The result is the same! 
Only NMEA works at 9600 speed and UBLOX does not work. How so? Is there a chance that the code fix I indicated above will be revised?

[image]

blckmn, 2026-09-27:
We're closing this as part of a clean-up of issues more than a year old. Sorry it didn't get a resolution here, that's on us. If it's still relevant on the latest release, please open a new issue with current details and link back here.

closed 2026-09-27 by blckmn