betaflight/betaflight #14249: Add support for drone category IMU's from TDK

haslinghuis, 2025-02-09:
### Is your feature request related to a problem? Please describe

N/A

### Describe the solution you'd like

ICM-40609-D and ICM-45605

### Describe alternatives you've considered

N/A

### Other information

https://product.tdk.com/en/search/sensor/mortion-inertial/imu/list#ref=characteristic&7market_s%5B2%5D%5B%5D=74002010&psts%5B%5D=0&psts%5B%5D=100&_l=20&_p=1&_c=pure_status-pure_status&_d=0

sugaarK, 2025-02-19:
Is there any benefit to adding these? both the BMI and 42688 gyros took a lot of time and work to get right thanks to missing/poor documentation..  adding gyros to our system is not a trivial process. look at the LSM gyros they are totally not useable yet..

Quick-Flash, 2025-02-19:
> * ICM40609, ICM45605 are specific targeted for drone applications

Not sure target is the right word, more like advertised for drone applications.
The ICM40609-D has twice the noise of the ICM42688-P, but has a higher accelerometer range (probably the "drone" centric part).

The ICM45605  is noisier than the ICM42688-P, and runs at 6.4k. Not sure what makes it drone centric.
ICM45605 and ICM45686 have "BalancedGyroscope Architecture" which is supposed to help the IMU deal with very high frequency oscillations (30khz oscillations) on AIO boards certain PWM frequencies of the ESC can actually interfere with the internal clock frequency of the IMU and cause really bad readings. The IMU oscillation frequency and the vibration interfere. So the ICM45605 and ICM45686 might be better in this regard.

The ICM45605 and ICM45686 may be a bit better at handling temperature drift than the ICM42688-P.

I've tested a few of these IMU at work, and they seem to perform relative to each other just how their datasheets claim. I don't have the equipment to test if their claims are valid, just that in comparison to each other the differences in their claims make sense.
https://invensense.tdk.com/products/motion-tracking/6-axis/

haslinghuis, 2025-02-19:
Forgot to mention we recently added IIM42653 IMU in https://github.com/betaflight/betaflight/commit/7a44f1bdce530e0dbf36c3a6acc794914bd3f790

Quick-Flash, 2025-02-19:
> Forgot to mention we recently added IIM42653 IMU in [7a44f1b](https://github.com/betaflight/betaflight/commit/7a44f1bdce530e0dbf36c3a6acc794914bd3f790)

Kiss ultra uses the IIM42652, so perhaps the IIM42653 will be as good or better than the ICM42688-p.

Quick-Flash, 2025-02-19:
It would be neat to make an FC that allows you to swap the IMU for some IMU testing.