betaflight/betaflight #14253: Flipover mode issue

mituritsyn, 2025-02-11:
### Describe the bug

Flip over after crash mode doesn't work. Arm tumbler behaves weirdly:
at first activation flip mode just does not work
after deactivation and reactivation the arm is activated
so every odd arm just doesn't do anything, an even arm results in a reqular arm and the quad can take off.
Video:
https://drive.google.com/file/d/1RL24wQxJ2g_TkS6JPppxWVak-eesoojl
DVR( you can see switch statuses and arming state):
https://drive.google.com/file/d/1tecuckWXAZgTJILwkP5yzZh-gdxAdGx_/view?usp=drive_link

### To Reproduce

flash 4.6 master

### Expected behavior

After arm in flip over mode I can control motors reverse direction with right stick

### Support ID

```txt
5825cc37-b82c-4caa-81b4-b1685ed92540
```

### Flight controller

HW f722 conv

### Other components

HW esc flashed to Am32

nerdCopter, 2025-02-11:
@ctzsnooze

nerdCopter, 2025-02-11:
Do you use a pre-arm switch? if yes, please try with `set prearm_allow_rearm = ON`,`save`. Otherwise i'll defer to other devs.

mituritsyn, 2025-02-11:
> Do you use a pre-arm switch?

nope

mituritsyn, 2025-02-13:
new data. flip mode is not active if the quad is not upside down, for this reason nothing happened in odd arms. there is logic in this when using the function directly. but it can be extremely useful when you are stuck on a tree and then the copter is not necessarily upside down, but it would be useful to be able to use the flip mode. the problem with a regular arm at even numbered times is relevant.

mituritsyn, 2025-02-13:
tested on other setup: 4.5 with blheli32 works as expected,  4.6 master - each even arm lead to regular arming

nerdCopter, 2025-02-13:
can you clarify " `each even arm` "?  do you mean even versus odd toggles?

mituritsyn, 2025-02-13:
> can you clarify " `each even arm` "? do you mean even versus odd toggles?

yes, even and odd arm stick activations

mituritsyn, 2025-02-13:
@ctzsnooze could you please take a look
```
$ git bisect good
7156dc84a39f0068b45d6afa82ccd3afa6fc6cb6 is the first bad commit
commit 7156dc84a39f0068b45d6afa82ccd3afa6fc6cb6
Author: ctzsnooze <chris.thompson@sydney.edu.au>
Date:   Sat Oct 5 07:58:33 2024 +1000

    Update turtle / crashflip mode (#13905)
```

nerdCopter, 2025-02-13:
please test with modes-tab beeper removed (or disable dshot beacons)

mituritsyn, 2025-02-13:
> please test with modes-tab beeper removed (or disable dshot beacons)

same, disabling dshot beacons doesn't affect the issue

nerdCopter, 2025-02-13:
thank you, i was following the code and noticed some toggles associated with beeper and switch combinations -- unfortunate it was not so simple.

can you test the non-legacy crashflip? i.e. `set crashfliprate = XX` where XX is some value 35 or higher.

nerdCopter, 2025-02-13:
i dont know why this `if-statement` changed. maybe @ctzsnooze can enlighten. or maybe it has nothing to do with the issue, i'm just trying to follow the code.

[image]

mituritsyn, 2025-02-13:
> `set crashfliprate = XX` where XX is some value 35 or higher.

`set crashfliprate = 80 `

with no changes

nerdCopter, 2025-02-13:
interesting comment https://github.com/betaflight/betaflight/pull/13905#discussion_r1954773744 

i was looking at it firstly as well, but reading the code-comments, it seemed okay.
However, it does lead to questions:
- What triggers `updateArmingStatus` function? Because, at the noted code, if the AUX is ON (BOXCRASHFLIP), then the `else-block` _should_ trigger.
- However, what substantiates `!crashFlipSuccessful()` if legacy crashflip is used.

ref: 

[image]

blckmn, 2026-09-27:
This has since been addressed in Betaflight, so we're closing it as part of a clean-up of issues more than a year old. If the latest release doesn't cover what you were after, please open a new issue and link back here.

closed 2026-09-27 by blckmn