betaflight/betaflight #14262: Betaflight Midair Restart Barometer

krzysztof176, 2025-02-16:
### Is your feature request related to a problem? Please describe

I noticed that when performing a midair restart, my Betaflight resets the altitude to zero while still in the air. As I descend, it then shows a negative altitude.

### Describe the solution you'd like

Having an option to reset the altitude when switching batteries instead of during arming would be great.

### Describe alternatives you've considered

N/A

nerdCopter, 2025-02-16:
when one restarts any program/server/service, all variables re-initialize, no?

krzysztof176, 2025-02-16:
The drone is still powered on, and everything else is functioning properly. If I can rearm the drone midair, it would be great to maintain altitude value and continue the flight without any issues. I understand that, by design, everything gets reinitialized, but it would be helpful to have an option to skip the altitude reset.

ledvinap, 2025-02-17:
What is the reason for midair restart? 
And is it even possible? Initial calibration should fail and prevent assuming.

krzysztof176, 2025-02-17:
One possible reason for a midair restart could be to conserve battery. You can rearm if the small_angle is set to 180 and the throttle is lowered before rearming. I noticed this issue when I accidentally disarmed and then turned the motors back on. I wouldn’t do this frequently, but it would be a useful feature to have.

In airplanes, when the engine turns off, they don’t lose altitude information, so a drone should behave similarly and retain its altitude data during a midair restart.

haslinghuis, 2025-02-17:
IMO baro should retain altitude independent on arm status while powered. Tried to find where it would be reset.

ledvinap, 2025-02-18:
@haslinghuis : maybe it could be handled together with gps home latch - using separate start-of-flight event - generated on all arm events or only on first one

krzysztof176, 2025-02-21:
Here is a video of the negative altitude after rearming.
https://drive.google.com/file/d/11eL3w_EHh-k5P-0KlBznL8aTwXFBB2rh/view?usp=sharing

krzysztof176, 2025-03-24:
Just following up—would someone be able to implement this?

closed 2025-05-02 by github-actions[bot]