betaflight/betaflight #14264: Flying in Wrong Direction With Fixed Altitude GPS Rescue

krzysztof176, 2025-02-16:
### Describe the bug

I'm using Betaflight 4.5 and have set GPS Rescue to return at a fixed altitude of 100m. However, when I ascended to around 200m and activated GPS Rescue, the drone flew in the wrong direction until it reached the set altitude, which is quite problematic.

### To Reproduce

1) Set a fixed altitude.
2) Ascend beyond the fixed altitude in the direction opposite to home.
3) Activate GPS Rescue.
4) The drone flies in the wrong direction.

### Expected behavior

Begin returning or hold position before reaching the set altitude.

### Support ID

```txt
# support date
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# 
Entering CLI Mode, type 'exit' to return, or 'help'

# AL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SOFTSERsource SPI_SDI 3 C11
resource SPI_SDI 4 NONE
resource SPI_SDO 1 A07
resource SPI_SDO 2 B15
resource SPI_SDO 3 C12
resource SPI_SDO 4 NONE
resource ESCSERIAL 1 NONE
resource CAMERA_CONTROL 1 A00
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C01
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource SDCARD_CS 1 D02
resource SDCARD_DETECT 1 NONE
resource PINIO 1 C09
resource PINIO 2 NONE
resource PINIO 3 NONE
resouB04: TIM3 CH1 (AF2)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer B00 AF2
[... 468 lines cut ...]
set launch_control_gain = 40
set thrust_linear mplified_i_gain = 100
set simplified_d_gain = 100
set simplified_pi_gain = 100
set simplified_dmax_gain = 100
set simplified_feedforward_gain = 100
set simplified_pitch_d_gain = 100
set simplified_pitch_pi_gain = 100
set simplified_dterm_filter = ON
set simplified_dterm_filter_multiplier = 100
set tpa_mode = D
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_low_rate = 20
set tpa_low_breakpoint = 1050
set tpa_low_always = OFF
set ez_landing_threshold = 25
set ez_landing_limit = 15
set end the command batch
batch end
```

### Flight controller

SpeedyBee V3 F7 Flight Controller Stack: 30x30 Drone FC Stack

krzysztof176, 2025-02-17:
Here is a video of the issue.
https://drive.google.com/file/d/1NWRGcLjjkyxolVNGwh_qAXHFvCoTjaeG/view?usp=sharing

anti-vaxxer, 2025-03-23:
I see what you mean, the distance to home continued to increase after you engaged GPS rescue.
It isn't intentionally flying the wrong way.
It's because when GPS rescue is engaged, the copter enters angle mode to level out, and then ascends/descends to `GPS_RESCUE_RETURN_ALT`, and Betaflight doesn't use position hold during that phase, so any prior speed you have (or wind) will make the copter drift until it reaches the altitude.

krzysztof176, 2025-03-23:
It would be beneficial to introduce a new setting called "Fixed Altitude Horizontal Position" with the following options: "N/A" (default), "Hold", and "Return to Home".

N/A – The drone does not attempt to correct its position, allowing wind or other factors to influence its movement.

Hold – The drone maintains its horizontal position while ascending or descending until it reaches the target altitude.

Return to Home – The drone immediately begins returning to its home point, even if it has not yet reached the desired altitude.

nerdCopter, 2025-03-24:
@ctzsnooze , please see yesterday's comments.

ctzsnooze, 2025-05-03:
4.6 does better I think. 

If there’s no Mag and the heading has not been established by prior forward flight, even 4.6 will need to fly in a straight line for about 60-100m

blckmn, 2026-09-27:
This has since been addressed in Betaflight, so we're closing it as part of a clean-up of issues more than a year old. If the latest release doesn't cover what you were after, please open a new issue and link back here.

closed 2026-09-27 by blckmn