betaflight/betaflight #14266: Gps Rescue Ascend/Descent Rate Base on Different Altitude

krzysztof176, 2025-02-17:
### Is your feature request related to a problem? Please describe

I adjusted my descent rate to be more aggressive, and I noticed that the landing wasn’t smooth.

### Describe the solution you'd like

To allow the configuration of different ascent and descent rates based on altitude.

Example:
- Above 100m: descent rate of 20 meters/second
- Between 25m and 100m: descent rate of 10 meters/second
- Below 25m: descent rate of 1.5 meters/second

The altitude ranges should also be configurable, allowing for any number of custom ranges.

### Describe alternatives you've considered

Simply reduce the descent rate to a lower value.

Quick-Flash, 2025-02-18:
This works well if you know where the ground is. PX4 uses this sort of logic to reduce descent and even horizontal max speeds as you approach the ground. Unfortunately we often don't know where the ground is. GPS and Barometer drift in the vertical direction, and you don't always land near your feet. For any flight that doesn't land near where it took off we have no real idea where the ground is.

krzysztof176, 2025-02-21:
I understand that we don't know exactly where the ground is, but that shouldn't prevent us from descending faster at higher altitudes and then slowing down at lower altitudes. I think this would be a great feature and significantly enhance the overall flight experience. By allowing users to configure the descent rate, we could make the system more flexible, enabling different descent or ascent rates for various altitudes. This would provide greater control and customization, making the flight smoother and more efficient based on user preferences.

krzysztof176, 2025-03-23:
Would this be possible to accomplish?

anti-vaxxer, 2025-03-23:
According to the [wiki](https://betaflight.com/docs/wiki/guides/current/GPS-Rescue-v4-5) the current implementation already includes a dynamic descent rate, but it's modest.
[image]
[image]

anti-vaxxer, 2025-03-23:
> Just to clarify, whatever value I set for "GPS_RESURE_DESCENT_RATE", it only applies to the landing phase and not the initial descent, correct? If that's the case, the initial descent rate is up to 2× "GPS_RESURE_DESCENT_RATE", right? 

Yes sounds right based on the wiki.

> If so, would it be possible to make the multiplier configurable, for example, allowing 5× instead of just 2×?

I guess so, but I'm not a developer.

I'm with you though it would be nice to have ability to choose descent rates based on altitude.
Because if for example you're at a very high altitude, then if the descent rate is too slow it may take too long to descend and the battery may even run out, in such cases you'd want a high descent rate to lose most of the altitude quickly, then have it switch to a slower rate at lower altitude.

krzysztof176, 2025-03-24:
@anti-vaxxer 
Do you know how to get a developer engaged in this?

Quick-Flash, 2025-03-25:
> [@anti-vaxxer](https://github.com/anti-vaxxer) Do you know how to get a developer engaged in this?

Do it yourself, get a developer excited about this, or pay someone.

krzysztof176, 2025-03-25:
@Quick-Flash 
So I can develop any feature I want and release it without needing approval, or is there an approval process for features? I just don’t want to waste time developing something that might not get released.

Quick-Flash, 2025-03-25:
You can develop it and have it work on your own fork. You can then pull request it and it may be merged. It likely would be if coded to BF standards.

closed 2025-05-02 by github-actions[bot]