betaflight/betaflight #14274: GPS: ublox spoofing indicator

anti-vaxxer, 2025-02-20:
### Is your feature request related to a problem? Please describe

When a GPS receiver provides coordinates that are wrong, spoofing can be the culprit.
During spoofing the satellite count can still be good, so the user has no idea that it's happening, causing the user to have false trust in the GPS.
It can be dangerous if for example GPS Rescue is initiated, the quad will fly in the wrong direction and may injure people.

### Describe the solution you'd like

ublox GNSS receivers have spoofing detection:
UBX-NAV-STATUS (0x01 0x03)
spoofDetState
there are four detection states:
0: Unknown or deactivated
1: No spoofing indicated
2: Spoofing indicated
3: Multiple spoofing indications

[image]
https://content.u-blox.com/sites/default/files/u-blox-M10-SPG-5.10_InterfaceDescription_UBX-21035062.pdf#%5B%7B%22num%22%3A1744%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C59.527%2C198.913%2Cnull%5D

I propose for Betaflight to take advantage of it and if spoofing is detected an OSD message be displayed notifying the user about it.

### Describe alternatives you've considered

ㅤ

KondratokGK, 2025-05-05:
I think it's really useful feature

jawz101, 2025-08-08:
I'm all for using as much of u-blox's features if betaflight is going to autconfigure whatever is programmed on the gps module.  Personally, I'd like to know Betaflight uses a bunch of u-Blox's features if they are there.