betaflight/betaflight #14277: Kakute Flight Controller Corrupts Blackbox files

mituritsyn, 2025-02-20:
### Describe the bug

My friend experiencing an issue with Blackbox logging on his KAKUTEH7MINI flight controller. The recorded Blackbox files contain corrupt headers, which include unreadable characters and duplicated blocks. This results in logs that are impossible to parse correctly.

`name:loopIteration,time,axisP[0],axisP[1],axisP[2],axisI[0],axisI[1],axisI[2],axisD[0],axisD[1],axisF[0],axisF[1],axisF[2],rcCommand[0],rcCommand[1],rcCommand[2],rcComm,axisF[1],axisF[2],rcCommand[0],rcCommand[1],rcCommand[2],rcCommQ p                           aÓâQ^ ãvž¹À9«¥ÖMåív_ÅÃPrb¬and[3],setpoint[0],setpoint[1],setpoint[2],setpoint[3],gyroADC[0],gyroADC[1],gyroADC[2],gyroUnfilt[0],gyroUnfilt[1],gyroUnfilt[2],debug[0],debug[1],debug[2],debug[3],debug[4],debug[5],debug[6],debug[7],motor[0],motor[1],motor[2],motor[3],eRPM[0],eRPM[1],eRPM[2],eRPM[3]`

The issue arose after upgrading from 4.5.0 dev build to 4.5.1 stable

from old log:
```
H Firmware revision:Betaflight 4.5.0 (515596328) STM32H743
H Firmware date:Apr  6 2024 05:07:28
```

### To Reproduce

1. Flash the latest stable version of Betaflight on a KAKUTEH7MINI flight controller.
2. Enable Blackbox logging and perform a test flight.
3. Download and analyze the Blackbox log file.

### Expected behavior

The Blackbox header should be correctly structured, pasable with no corruption.

### Support ID

```txt
Id: "958bca75-e346-4742-b7ff-273fe41d4122"
```

### Flight controller

KAKUTEH7MINI

demvlad, 2025-02-21:
That mean, it was not random issue.
rcCommand[3] caption is corrupted only, the all other data is good.

mituritsyn, 2025-02-21:
added a log from 4.6. not parsable as well but corruption is not in rcCommand field

ledvinap, 2025-02-21:
The corruption starts after first 0x100 bytes (`,rcComm,axisF[1]`, in all log files)

demvlad, 2025-02-21:
> The corruption starts after first 0x100 bytes (`,rcComm,axisF[1]`, in all log files)

0x100 .... 0x150

[image]

demvlad, 2025-02-21:
Has lower log rate good results?

haslinghuis, 2025-02-21:
Also is the issue persistent on 4.5.2-dev or 4.6.0-dev ?

mituritsyn, 2025-02-21:
> Also is the issue persistent on 4.5.2-dev or 4.6.0-dev ?

still there on 4.6-dev(last log in gdrive), 4.5.2 wasn't tested

mituritsyn, 2025-02-21:
> Has lower log rate good results?

I'll ask to test, but 1600 is lowest suitable for filter tuning and it is h7 FC. There was no issue in earlier version (4.5-dev) even with 3200 log rate

SteveCEvans, 2025-02-21:
Please provide the `status` output. What size is the FLASH. Early parts on these FCs were counterfeit, small capacity, and slow.

demvlad, 2025-02-21:
>  @demvlad managed to manually fix one of the logs using Notepad++ on Windows. 

The caption is destroyed very hard. I've tryed to restore, but i can not do this.

SteveCEvans, 2025-02-22:
Thanks. I’ll do some testing with H7 and also with that FLASH device to try to reproduce.

mituritsyn, 2025-03-24:
After flash erasing, logs are not corrupted anymore. Therefore, it could be a flash issue or related to cases where the flash is almost full.