betaflight/betaflight #14281: MAX7456 invalid SPI mode

chemicstry, 2025-02-26:
### Describe the bug

We are developing a replacement for MAX7456 OSD chip and noticed that betaflight uses SPI mode 3 (CPOL = 1, CPHA = 1), which is wrong. According to MAX7456 datasheet, this chip uses SPI mode 0 (CPOL = 0, CPHA = 0), just like SD cards.

I'm not sure why this is working as is, probably just a lucky coincidence, because MAX7456 does not seem to support automatic SPI mode negotation.

For comparison, INAV uses correct mode 0.

### To Reproduce

Run betaflight

### Expected behavior

MAX7456 should use SPI mode 0

### Support ID

```txt
Not applicable
```

### Flight controller

Custom

SteveCEvans, 2025-02-26:
Which version of Betaflight were you using when you tested this? See https://github.com/betaflight/betaflight/pull/14142 which shows that the SD configuration was incorrect and causing issues.

From https://www.analog.com/media/en/technical-documentation/data-sheets/max7456.pdf you'll see that SCLK must be high before CS is asserted as per the tCSH0 timing. This corresponds to CPOL=1.

[image]

chemicstry, 2025-02-27:
I was testing using `4.5.1` so no, I did not have that change yet. But it only affects SD card driver, not the MAX7456.

Also, this is what I get from ChatGPT:
```
- What SPI mode (CPOL, CPHA) does SD card use?

SD cards use SPI Mode 0 (CPOL = 0, CPHA = 0) or SPI Mode 3 (CPOL = 1, CPHA = 1), depending on the implementation.

However, SPI Mode 0 (CPOL = 0, CPHA = 0) is the most commonly used and recommended mode for communication with SD cards.
```

I couldn't find the proper SD card specification, because it is behind paywall. The free simplified specification does not have SPI timing diagrams.

The MAX7456 datasheet is a bit confusing on this part. I believe tCSH0 refers to time between last byte of previous transaction and the CS assertion of the next one. All the figures following fig. 14 show clock as idle low.

I will do some more investigation and report back.

nerdCopter, 2025-02-27:
to assist other devs and have the latest code, please test with 4.5.2 and/or 4.6

[image]

closed 2025-04-06 by github-actions[bot]