betaflight/betaflight #14288: Landing and GPS Rescue

Denethor85, 2025-03-04:
### Is your feature request related to a problem? Please describe

Currently if you do not set GPS Rescue mode (which also includes landing and automatic engine shutdown) you can choose the simple landing mode. The behavior of Simple Landing mode without GPS Rescue is unfortunately not consistent with GPS Rescue landing.

Also if there is a problem in the GPS Rescue the drone falls to the ground without control, maybe in some situations it would be better to have a more controlled landing 

### Describe the solution you'd like

It would be nice if we could implement 2 things for the future:

1) Make the Simple Landing mode without GPS Rescue consistent in behavior and the way it acts like GPS Rescue landing, i.e., with automatic shutdown of the motors as soon as the drone touches the ground, without the need to manually set the time to leave the motors on (which risks burning esc and motors)

2) The ability to choose and set Betaflight to go into landing mode if something in GPS Rescue fails (lack of satellites, etc.) instead of crashing to the ground, then allowing the choice by adding the ability to start GPS Rescue landing.

Thanks

### Describe alternatives you've considered

None

AirCombat, 2025-04-15:
I think this would come under the work for Optical Flow (controlled landing) maybe even, a final stage of failsafe

I think a seperate feature request covering fallback for GPS rescue would be good

ctzsnooze, 2025-05-03:
This can be done fairly easily using altitude control alone. The quad would drift. The quad would need a Baro or a GPS. If it had a GPS and the IMU was oriented then it could use position hold to not drift. I was going to do this (leveraging the new altitude and position hold functions) but didn’t get around to it.

blckmn, 2026-09-27:
This has since been addressed in Betaflight, so we're closing it as part of a clean-up of issues more than a year old. If the latest release doesn't cover what you were after, please open a new issue and link back here.

closed 2026-09-27 by blckmn