betaflight/betaflight #14294: BZGNSS BZ-181 not workinh with Betaflight 4.5.1 but OK with 4.5.0

jpbaboo, 2025-03-11:
### Describe the bug

HI
When configuring  BZGNSS BZ-181 with betaflight 4.5.1 we got the message ''Searching Satelite' and 'waitin for date when saving the configuration..
When downgrading to 4.5.0.. evrithing is fine..
Hope this help

### To Reproduce

Configuring  GPS  BZ-181 with UBLOX... Betaflight 4.5.1 stuck  when searching satelite
With 4.5.0 all is  OK

### Expected behavior

Waiting for GPS to be configured

### Support ID

```txt
Sorry   
2025-03-11 @08:04:15 -- Mode CLI détecté

2025-03-11 @08:07:38 -- Échec de construction du serveur : HTTP 400

2025-03-11 @08:07:39 -- Échec de construction du serveur : HTTP 400
```

### Flight controller

F405 V4 Speedy bee

### Other components

 BZGNSS BZ-181

haslinghuis, 2025-03-11:
- WS M181 works on SPBE F405V4
- Get support ID from CLI tab and post here
- Enable Auto Config if not enabled already
- Try 4.5.2-dev
- Try 4.6.0-dev

jpbaboo, 2025-03-21:
Hi  Thks for your answer..
System Engeener in ComputerSciences.. i'm afraid  to install Dev Version  
As i said all is fine in 4.5.0.. i will wait to upgrade to 4.5.2 or 4.6.0 when avalaible
See U

closed 2025-03-21 by jpbaboo