betaflight/betaflight #14298: Betaflight crash_recovery gui for configurator

danielLublinsky, 2025-03-13:
### Is your feature request related to a problem? Please describe

Recently I discovered the crash_recovery feature that exists in the cli, it is surprising that this feature isn't supported in the gui and it could become more available by adding it as a option.

### Describe the solution you'd like

I would like to see at least the option to set the mode (on,off,beep,disarm) in the configuration tab, but adding basic tuning options could also be nice.

### Describe alternatives you've considered

This option could be added to the failsafe tab instead, this is your call.

### Other information

If this option is in your interest, I would be more then happy to implement it myself and contribute to the development.

haslinghuis, 2025-03-14:
- crash recovery is part of the pid profile so any changes would need to go into pid tuning tab.
- perhaps we also discuss moving `crash_recovery` out of the `profile` and use `master` instead ?

danielLublinsky, 2025-03-16:
Crash recovery belongs in the master settings because it’s not something that gets changed often and doesn’t really depend on the PID profile.

thespova, 2025-11-08:
> Crash recovery belongs in the master settings because it’s not something that gets changed often and doesn’t really depend on the PID profile.

I agree with this. Its a bit annoying to have to remember to go make changes on each profile. Perhaps there is a pid setting that makes it so they need to be profile specific but it does not seem like it to me.

blckmn, 2026-09-27:
We're closing this as part of a clean-up of issues more than a year old. Sorry it didn't get a resolution here, that's on us. If it's still relevant on the latest release, please open a new issue with current details and link back here.

closed 2026-09-27 by blckmn