betaflight/betaflight #14301: make RP2350 fatal error: io_def_generated.h : No such file or directory

tigrajack, 2025-03-15:
### Describe the bug

make RP2350 
In file included from ./src/main/drivers/io.h:40,
                 from ./src/main/drivers/sdcard.c:29:
./src/main/drivers/io_def.h:53:10: fatal error: io_def_generated.h: No such file or directory

### To Reproduce

sudo apt install build-essential libblocksruntime-dev libtool git curl clang-18
make arm_sdk_install
make configs
git pull

make RP2350

### Expected behavior

perl def_generated.pl not work for PICO RP2350

### Support ID

```txt
NO CLI
```

### Flight controller

RP2350

nerdCopter, 2025-03-25:
> When do you plan to solve the problem? You have a logo "Raspberry Pi" on your site, but the any RPi boards does not compile. This looks like consumer fraud.

it is a work-in-progress not ready yet. 
it will occur when developers who volunteer their time are able. 
https://github.com/betaflight/betaflight/pulls?q=is%3Apr+RP2350+

blckmn, 2025-04-02:
Closing this issue as RP2350 is a work in progress, and is not yet in market.

closed 2025-04-02 by blckmn