betaflight/betaflight #14305: Support for multiple free axis motors

goofyseeker311, 2025-03-17:
### Is your feature request related to a problem? Please describe

if there is not already, multiple free (fixed) axis motor placement logic would be useful.

### Describe the solution you'd like

not just flat plane upwards motor axis placement of a quad.
mostly for perpendicular/3d cross pattern placement of motors axis.

### Describe alternatives you've considered

none yet, custom code implementation.

Quick-Flash, 2025-03-17:
you can already handle this in the mixer
however, current BF logic will only handle using these motors for pitch/roll/yaw (any combination of those) and not for translation

goofyseeker311, 2025-03-17:
all combinations of direction and translation are essential.

ledvinap, 2025-03-17:
@goofyseeker311 : IIRC you can use thrust in motor mixer. 
There is no translation signal in current BF control loop, only rotations + thrust. Changing it would be really hard (and useless for most users)

goofyseeker311, 2025-03-17:
I guess I'll do "up" as forward as usual, all the other axis as rotation control.

papaxiclas, 2025-04-17:
Hello! I am currently having the same need but, for example, for translation I would only need the "add" off the value of a radio channel to the specific motor.
I think it is just adding that channel value to the power variable before it is sent to the ESC. 
Is this possible somehow?