betaflight/betaflight #14308: Meteor 65 pro elrs 2022 Smart Audio Problem on 4.5.1

Hayk25, 2025-03-18:
### Describe the bug

Meteor 65 pro elrs 2022 Smart Audio Problem on 4.5.1. When I install 4.5.0 smart audio works, when 4.5.1 it stops working and the device is not distributed. Tried on 8 different drones, the problem is the same.

### To Reproduce

Meteor 65 pro elrs 2022 Smart Audio Problem on 4.5.1. When I install 4.5.0 smart audio works, when 4.5.1 it stops working and the device is not distributed. Tried on 8 different drones, the problem is the same.

### Expected behavior

4.5.1 or 4.6 smart audio fix forbetafpv meteor 65 pro elrs drone.

### Support ID

```txt
Smart audio
```

### Flight controller

4.5.1

haslinghuis, 2025-03-18:
- Should be fixed in firmware 4.5.2-dev in [28e9df1](https://github.com/betaflight/betaflight/commit/28e9df19d33894ec246ef8af641197f364d22631)
- Support ID can be generated using button in CLI tab in configurator.

closed 2025-04-27 by github-actions[bot]

haslinghuis, 2025-04-27:
Duplicate of #14316

stalkerok1993, 2026-02-16:
Unfortunately SmartAudio still doesn't work for Meteor65 Pro starting from BTFL 4.5.1 (tried 4.5.2 and latest 4.5.3 & 2025.12.2).