betaflight/betaflight #14312: Gyro loop unstable on F405 with 24 MHz HSE

ot0tot, 2025-03-19:
### Describe the bug

Board functions normally when using HSI, but when `system_hse_mhz` is set to 24 the gyro loop becomes unstable. 

`CPU:0%, cycle time: -78089924, GYRO rate: 0, RX rate: 49, System rate: 0`

Video of  issue on manufacturer Discord: https://discord.com/channels/868013470023548938/1293122863724757064/1351898414165790823

### To Reproduce

```
set system_hse_mhz = 24
save
```

### Expected behavior

System should function normally using 24 MHz oscillator. 

### Support ID

```txt
# X-Request-Id: 5e58caa4-f83e-47cb-b819-25a9ce49f756

###

# # Problem description

# # new support id , requested by BF Team.

# ###

# version
# Betaflight / STM32F405 (S405) 4.5.1 Mar 19 2025 / 10:06:16 (77d01ba3b) MSP API: 1.46
# config rev: 0f78778
# board: manufacturer_id: BLDY, board_name: BOTWINGF405

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=51degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3450, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P
BUILD KEY: f6ebd76788d54c087d5f9d5394d5bfb3 (4.5.1)
System Uptime: 65 seconds, Current Time: 2025-03-19T07:07:29.231+00:00
CPU:0%, cycle time: -78089924, GYRO rate: 0, RX rate: 49, System rate: 0
Voltage: 812 * 0.01V (2S battery - OK)
I2C Errors: 1
FLASH: JEDEC ID=0x00ef4018 16M
GPS: NOT ENABLED
Arming disable flags: ANGLE LOAD CLI MSP

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump master

# version
# Betaflight / STM32F405 (S405) 4.5.1 Mar 19 2025 / 10:06:16 (77d01ba3b) MSP API: 1.46
# config rev: 0f78778

# start the command batch
batch start

board_name BOTWINGF405
manufacturer_id BLDY
[... 917 lines cut ...]
    CH2 : MOTOR 1
TIM2:
    CH2 : LED_STRIP
TIM3:
    CH3 : MOTOR 4
    CH4 : MOTOR 3
TIM4:
    CH2 : PPM
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

#
```

### Flight controller

BOTWINGF405 dev sample

ledvinap, 2025-03-19:
Can you check frequency for HSI? (MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=51degC)

ot0tot, 2025-03-19:
> Can you check frequency for HSI? (MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=51degC)

Support ID using HSI:
```
# # Problem description

# # Crystal of F405 Botwing,Requested by betaflight

# ###

# version
# Betaflight / STM32F405 (S405) 4.5.1 Mar 19 2025 / 10:06:16 (77d01ba3b) MSP API: 1.46
# config rev: 0f78778
# board: manufacturer_id: BLDY, board_name: BOTWINGF405

# status
MCU F40X Clock=168MHz (PLLP-HSI), Vref=3.30V, Core temp=45degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3450, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
BUILD KEY: f6ebd76788d54c087d5f9d5394d5bfb3 (4.5.1)
System Uptime: 112 seconds, Current Time: 2025-03-19T11:28:25.051+00:00
CPU:45%, cycle time: 127, GYRO rate: 7874, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - INIT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
GPS: NOT ENABLED
Arming disable flags: RXLOSS ANGLE CLI MSP

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump master

# version
# Betaflight / STM32F405 (S405) 4.5.1 Mar 19 2025 / 10:06:16 (77d01ba3b) MSP API: 1.46
# config rev: 0f78778

# start the command batch
batch start

board_name BOTWINGF405
manufacturer_id BLDY

# name: -

# resources
resource BEEPER 1 B00
resource MOTOR 1 A09
resource MOTOR 2 A08
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource PPM 1 B07
resource LED_STRIP 1 B03
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
[... 891 lines cut ...]
Currently active Timers:
-----------------------
TIM1:
    CH1 : MOTOR 2
    CH2 : MOTOR 1
TIM2:
    CH2 : LED_STRIP
TIM3:
    CH3 : MOTOR 4
    CH4 : MOTOR 3
TIM4:
    CH2 : PPM
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

# 
```

nerdCopter, 2025-03-19:
although helpful, a full dump / buildkey is not a support ID.
[image]
[image]

ot0tot, 2025-03-19:
> although helpful, a full dump / buildkey is not a support ID. [image] [image]

These are the support ID outputs, not CLI dumps.
If you want the IDs:
HSI: `94477940-69c3-4858-a268-f0e39c223496`
HSE: `6c33c24d-fbae-4c51-ac2c-2a80d8a0171e`

SteveCEvans, 2025-04-07:
I can confirm this behaviour. The `SysTick_Handler()` is being called at the required 1ms interval, so the basic processor clock timing appear correct, however the gyro task is sometimes being called very sporadically.

[image]

However at other times it's almost correct but with a lot of jitter.

[image]

ledvinap, 2025-04-08:
Maybe SPI clock is set to wrong frequency with HSE?
Can you check HSE vs HSI clock settings against reference manual (or replicate settings in cubeMX)?
IIRC SysTick_Handler frequency is derived using HAL, so it should use actual hardware settings. But other peripheral clock calculation is done (at least partially) in BF, it may be wrong.

@SteveCEvans : Can you verity SPI clock speed?

Also, it would be interesting to check with crystal that has same frequency as HSI (16MHz?)

SteveCEvans, 2025-04-08:
The following loop is being used to test the accuracy of micros().

```
            while(1) {
            	currentTimeUs = micros();
            	if ((currentTimeUs % 100) < 10) {
		     pinioSet(0, 1);
            	} else {
		    pinioSet(0, 0);
            	}
            }
```

And it results in the following which case be seen to be three times as fast as it should be. This ratio is the same as 8 vs 24MHz, so we're onto something...

[image]

SteveCEvans, 2025-04-08:
See https://github.com/betaflight/config/pull/723#discussion_r2034087256

SteveCEvans, 2025-04-08:
> Maybe SPI clock is set to wrong frequency with HSE? Can you check HSE vs HSI clock settings against reference manual (or replicate settings in cubeMX)? IIRC SysTick_Handler frequency is derived [...]

The config file was incorrectly setting `SYSTEM_HSE_MHZ` to 8 when it should be 24.

SteveCEvans, 2025-04-08:
I suggest we remove the `system_hse_mhz` setting as the value is hardcoded as a constant defined by `SYSTEM_HSE_MHZ`, and `system_hse_mhz` doesn't fully control the clock control behaviour. Thoughts?

haslinghuis, 2025-04-08:
Agree it is hard coded as default. But could be changed in cli (or diff)

https://github.com/betaflight/betaflight/blob/d9f418dd06ccac4b8944966229ac033d004a04a4/src/main/cli/settings.c#L1676-L1678

https://github.com/betaflight/betaflight/blob/d9f418dd06ccac4b8944966229ac033d004a04a4/src/main/fc/init.c#L486-L490

SteveCEvans, 2025-04-08:
At the moment you have to set `system_hse_mhz` to a non zero value for status to report `HSE` is being used thus:

```
# status
MCU STM32F40X Clock=168MHz (PLLP-HSE), Vref=3.29V, Core temp=44degC
```

If left at zero `HSI` is used by default thus:

```
# status
MCU STM32F40X Clock=168MHz (PLLP-HSI), Vref=3.29V, Core temp=45degC
```

Interestingly looking at the full status output with `HSI` used we get a gyro rate of 8kHz.
```
# status
MCU STM32F40X Clock=168MHz (PLLP-HSI), Vref=3.29V, Core temp=45degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4484, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MSP (53 x 20)
System Uptime: 13 seconds, Current Time: 2025-04-08T22:35:08.168+00:00
CPU:41%, cycle time: 125, GYRO rate: 8000, RX rate: 0, System rate: 9
Voltage: 11 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
GPS: NOT ENABLED
Arming disable flags: RXLOSS ANGLE CLI MSP
```
However with `HSE` we see a higher figure. I'll look into this some more.
```
# status
MCU STM32F40X Clock=168MHz (PLLP-HSE), Vref=3.29V, Core temp=45degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4484, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MSP (53 x 20)
System Uptime: 22 seconds, Current Time: 2025-04-08T22:37:16.359+00:00
CPU:42%, cycle time: 124, GYRO rate: 8064, RX rate: 0, System rate: 9
Voltage: 11 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
GPS: NOT ENABLED
Arming disable flags: RXLOSS ANGLE CLI MSP
```

ledvinap, 2025-04-09:
@SteveCEvans:
- some startup/system_xx.c files redefine SystemCoreClockUpdate to take use correct HSE value (F4, G4, APM32F4)
- a lot of library code does use HAL_RCC_GetSysClockFreq (also system.c/cycleCounterInit), which uses hardcoded value - modification to SystemCoreClockUpdate is not enough

It is possible to fix drivers, but it won't be trivial.

For now, is it probably best to document situation and use system_hse_mhz as on/of switch:
- 0 = use HSI
- != 0 - replace with HSE_VALUE and enable HSE.

SteveCEvans, 2025-04-10:
Being able to set `system_hse_mhz` to a value which doesn't match `HSE_VALUE` is broken though. I'm thinking we would be better to have a `clk_src` setting which can take either `HSI` or `HSE` as parameters which will default to `HSE` if `SYSTEM_HSE_MHZ` is non-zero.

ledvinap, 2025-04-10:
@SteveCEvans : using system_hse_mhz will be (a bit) backward compatible and it will allow implementing runtime-customization later. It can also contain compiled-in HSE_VALUE, so mismatch will be a bit less likely.

blckmn, 2026-09-27:
This has since been addressed in Betaflight, so we're closing it as part of a clean-up of issues more than a year old. If the latest release doesn't cover what you were after, please open a new issue and link back here.

closed 2026-09-27 by blckmn