betaflight/betaflight #14316: SmartAudio does not work after upgrade to BT 4.5.2 on SpeedyBee F405 V4

mirktmirkt, 2025-03-22:
### Describe the bug

Hello,

I have SpeedyBee F405 V4 FC with TBS Unify Pro2 Nano 5G8 V1.1 VTX..

[image]

[image]

After upgrading BT to 4.5.2 and importing saved config file, SmartAudio stopped working:

[image]
[image]

Any ideas?

Regards,
mirkt

### To Reproduce

Upgrade BetaFlight to 4.5.2

### Expected behavior

SmartAudio working as expected :)

### Support ID

```txt
Id: "7c474f88-06f4-4ca5-aeb5-6212f63afce7"
```

### Flight controller

SpeedyBee F405 V4

### Other components

TBS Unify Pro2 Nano 5G8 V1.1

haslinghuis, 2025-03-22:
See also:
- https://github.com/betaflight/betaflight/pull/14246
- https://github.com/betaflight/betaflight/pull/14283

- IMO issues were introduced in https://github.com/betaflight/betaflight/pull/13771

Did it work in 
- 4.5.0
- 4.5.1

haslinghuis, 2025-03-22:
> make SPEEDYBEEF405V4 EXTRA_FLAGS="-D'BUILD_KEY=cbd11476e81895287b892f4bc9fa5e81' -D'RELEASE_NAME=4.5.2' -DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_AKK_SMARTAUDIO -DUSE_BATTERY_CONTINUE -DUSE_CAMERA_CONTROL -DUSE_DASHBOARD -DUSE_DSHOT -DUSE_EMFAT_AUTORUN -DUSE_EMFAT_ICON -DUSE_EMFAT_TOOLS -DUSE_ESCSERIAL_SIMONK -DUSE_FRSKYOSD -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_LED_STRIP_64 -DUSE_MAG -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIAL_4WAY_SK_BOOTLOADER -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERVOS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"

@mirktmirkt this does not make sense  - Only use defines needed.

Why are you including all these defines. `AKK_SMARTAUDIO` could cause this issue.

mirktmirkt, 2025-03-23:
Hello,

It was working fine in BTFL 4.5.1. I have never used BTFL 4.5.0 with this FC, but now I tried downgrading to 4.5.0, and SmartAudio works perfectly:
[image]

I never paid much attention to those OPTIONS, I tried 4.5.2 only with these OPTIONS, but SmartAudio doesn't work.
[image]

What else could I be doing wrong?

Regards,
mirkt

spaivaras, 2025-03-27:
Exactly the same issue with BetaFPV Air65 with 5in1 FC

CPU: STM32G473CEU6
VTX: uknown AV05B (42417)
SA: 2.0

Works fine with 4.5.0, 4.5.1
4.5.2 = SmartAudio Uknown

haslinghuis, 2025-03-27:
@spaivaras can you try adding `SMARTAUDIO_NOPULLDOWN` to custom defines before flashing ?

haslinghuis, 2025-03-27:
IMO we should revert https://github.com/betaflight/betaflight/pull/13771 and fixes needed so far. It worked for 7 years.

ledvinap, 2025-03-27:
@haslinghuis : https://github.com/betaflight/betaflight/pull/13663
Reverting https://github.com/betaflight/betaflight/pull/13771 will probably break some SA implementations again. We can break and fix different versions forever.

spaivaras, 2025-03-27:
Not sure if it helps anybody, but:

Managed to get it working on tag 4.5.2  with Air65 5in1

reverting only https://github.com/betaflight/betaflight/pull/13771 was not enough as it seems there where additional changes on[ io/vtx_smartaudio.c](https://github.com/betaflight/betaflight/pull/14246/files#diff-8a213843773adccbed4197a4e66e8faacd6ba388205446828ea864ef82faf785R500)

Using 
```
 make BETAFPVG473 EXTRA_FLAGS="-DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_DSHOT -DUSE_OSD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX -DUSE_NONCOMPLIANT_SMARTAUDIO"
```
(basically reverting sa changes from both PR's)
Finally got it working 

[image]

spaivaras, 2025-03-27:
Actually scratch that, the only thing that was needed was flag `USE_NONCOMPLIANT_SMARTAUDIO` for STM32G4 🤦 

@mirktmirkt You could also try it by adding `NONCOMPLIANT_SMARTAUDIO` to custom defines in firmware builder

nerdCopter, 2025-03-28:
> Actually scratch that, 

so `USE_NONCOMPLIANT_SMARTAUDIO` worked with released 4.5.2 @ `024f8e13d`?

spaivaras, 2025-03-28:
Yes for air65

[image]
[image]

Will try to test with stm32f4 (speedy bee 405 v4) but don't have a vtx at hand atm

Alexud4fd, 2025-03-29:
In version 4.5.2 it does not see VTX FX2 via SMARTAUDIO protocol on SPEEDYBEEF405V3, in version 4.5.1 everything works

nerdCopter, 2025-03-31:
@Alexud4fd , does this work or no?
[image]

LYNHQQ, 2025-04-01:
Using the AKK_SMARTAUDIO macro may cause some unexpected problems。

mirktmirkt, 2025-04-01:
> [@mirktmirkt](https://github.com/mirktmirkt) You could also try it by adding `NONCOMPLIANT_SMARTAUDIO` to custom defines in firmware builder

it worked, thank you!

mirktmirkt, 2025-04-01:
> [@Alexud4fd](https://github.com/Alexud4fd) , does this work or no? [image]

I can confirm, that this works with "SpeedyBee F405 V4" and "TBS Unify Pro2 Nano 5G8 V1.1"

haslinghuis, 2025-04-01:
Ooh - this is even needed for TBS hardware ?

@bkleiner

Alexud4fd, 2025-04-03:
On firmware 4.5.2 after enabling NONCOMPLIANT_SMARTAUDIO AKK VTX it worked completely
thanks

SentinelFour, 2025-04-13:
I am in the same boat. I use a TBS Unify Pro32 Nano and a Foxeer FC. Worked on 4.5.0. No on 4.5.2 and 4.6 no smart audio communication possible.

haslinghuis, 2025-04-14:
@ledvinap please check this issue

haslinghuis, 2025-04-14:
- Serial implementation has been improved in 4.6 which requires re-evaluating the need for additional defines.
- Instead of using defines perhaps we add a CLI parameter for firmware 4.6 to allow configuration.
- If possible like to remove `SMARTAUDIO_NOPULLDOWN` define too.

- https://github.com/betaflight/betaflight/pull/14342

haslinghuis, 2025-04-14:
@SentinelFour when flashing 4.6 development add this

[image]

- Change `vtx_serial_alt` parameter in CLI to switch behavior

ledvinap, 2025-04-15:
_this my personal opinion, I won't protest if it is handled differenty_

> * Serial implementation has been improved in 4.6 which requires re-evaluating the need for additional defines.

IIRC it was mostly unified - different CPUs are more consistent.

SA is very specific protocol (break is transmitted when idle), it is very hard to reason about it without trace of actual UART signals.

> * Instead of using defines perhaps we add a CLI parameter for firmware 4.6 to allow configuration.

I prefer NONCOMPLIANT define - user must invest a little bit of time when using buggy SA implementation. 

Only SA implementations adhering to official documentation should be supported with default configuration. It is possible that some UART implementations do not behave this way now - we should change that, fixing official SA (and possibly breaking noncompliant ones)

> * If possible like to remove `SMARTAUDIO_NOPULLDOWN` define too.
> * [Make SA prepend byte configurable using CLI #14342](https://github.com/betaflight/betaflight/pull/14342)

Technically it is quite easy to do so. But SA bugs are repeating again and again. Adding easy SA workaround will only lead to new suboptimal protocols/buggy protocol implementation in future. And more BF support issues.

ledvinap, 2025-04-16:
@SentinelFour : It's hard to tell without oscilloscope trace of SA pin.

It is also possible that you are  doing something wrong - #14342 with `vtx_serial_alt = ON` should not change anything on F7 (code defaults to ON on F7 before #14342)

SentinelFour, 2025-04-16:
Switched the command back to OFF:

[image]

Multiple power cycle; remained unchanged. 

Switched the command back to ON:

[image]

Multiple power cycle; remained unchanged. 

(I don't want to sound as if I know any better; not all - I have no clue what I am doing. Just hope that this helps to identify the root cause.)

ledvinap, 2025-04-16:
@SentinelFour : it won't work with `vtx_serial_alt = OFF`. But it should work if you do not include https://github.com/betaflight/betaflight/pull/14342 in build - select `latest (master)` as 'pull request or commit'

SentinelFour, 2025-04-16:
Can confirm, now it works. 
I tried 4.6 latest master previously... and there it did not work. So I indeed did something wrong back then or something has been changed in the latest master. If it was on me, then sorry for the fuzz. 

[image]

haslinghuis, 2025-04-16:
No worries - much improvements and refactoring efforts were made lately.

vkolotov, 2025-04-23:
I had the same issue with TinyHawk 2 woop/freestyle, this seems to be fixed in the 4.6:

[image]

However, I can't really test it as ERLS (crfs) is broken now there.

odddollar, 2025-04-27:
I can also confirm that I was having the same issue with the TBS Unify Pro32 Nano and a JHEMCU F405 NOXEv2. No smartaudio on 4.5.2, but on 4.5.1 it works perfectly. If this has been fixed in 4.6 then I might be staying on 4.5.1 until 4.6 comes out as a full release

haslinghuis, 2025-04-27:
@odddollar please confirm it works on 4.6.0

mattiamari, 2025-05-10:
Same issue here with MATEKF411 (DarwinFPV AIO) and DarwinFPV 5.8Ghz 40CH VTX. Smartaudio does not work (device ready = false) on both 4.5.1 and 4.5.2 but works fine on 4.5.0. `NONCOMPLIANT_SMARTAUDIO` has no effect.

haslinghuis, 2025-05-13:
@mattiamari should probably be fixed in 4.6.0. Please verify

In 4.5.2 - using softserial - could try with `USE_SMARTAUDIO_NOPULLDOWN` define

mattiamari, 2025-05-17:
> [@mattiamari](https://github.com/mattiamari) should probably be fixed in 4.6.0. Please verify
> 
> In 4.5.2 - using softserial - could try with `USE_SMARTAUDIO_NOPULLDOWN` define

Does NOT work on 4.6.0 out of the box. Does it need some custom defines?

haslinghuis, 2025-05-17:
@mattiamari add `#14342` in `select pull request or commit` before flashing

[image]

In CLI set this setting OFF when still not working:

```
set vtx_serial_alt = OFF
```

mattiamari, 2025-05-20:
@haslinghuis still not working with both `vtx_serial_alt = ON` and `OFF`

mattiamari, 2025-05-22:
@haslinghuis 
- 8db2b8b3-b755-4006-85ac-ba44e5a100a3
- no, just UART2
- not working on 4.5.2 with both AKK_SMARTAUDIO and NONCOMPLIANT_SMARTAUDIO

marcos-sm89, 2025-05-22:
As @spaivaras suggested, I added USE_NONCOMPLIANT_SMARTAUDIO custom define while flashing for

[image]

It solved the issue

[image]

Teramaks, 2025-06-21:
Had same issue, used AKK (SA FIX) and USE_NONCOMPLIANT_SMARTAUDIO while flashing and it helped. Rush Blade F7 V2 Analog flight controller

spaivaras, 2025-06-23:
> All - please also try 4.6 development build, no custom defines needed.

Just checked with BetaFPV Air65 and Air75 (both use the same BETAFPVG473 5in1 board) 4.6.0-dev SA works out of the box, no defines, no new settings changed. Build log: https://build.betaflight.com/api/builds/933d944a0067286c4a21a3f63561e124/log

Thank you 🙇‍♂

gpomykala, 2025-07-23:
USE_NONCOMPLIANT_SMARTAUDIO helped for air65 and BF4.5.2

thespova, 2025-08-02:
Using SA on an Emax NanoHawk AIO board "USE_NONCOMPLIANT_SMARTAUDIO" worked for 4.5.2
4.5.2 "USE_NONCOMPLIANT_SMARTAUDIO" worked and allowed smart audio to connect
4.6 Works with no custom defines on this board
Note: the AKK fix did not work with this board

haslinghuis, 2025-08-02:
No, 4.6 would put an end to this drama :)

closed 2025-09-10 by github-actions[bot]

fjulian79, 2025-11-30:
Same issue here on a MATEKF411RX (Tinyhawk S).
Yes, it's an old board, but that's what I had lying around while getting back into the hobby :-)

❌ 4.5.2 and 4.5.3 → not working
❌ 4.5.3 with USE_NONCOMPLIANT_SMARTAUDIO → still not working
✅ 4.5.1 and 2025.12.0-RC2 → working

I’m staying on 4.5.1 for now, as the whoop-specific presets do not apply error free on 2025.12.0-RC2.

In my opinion this issue should be reopened, as the problem is confirmed.
If it will be fixed in the final 2025.12.0 release (like it currently is in RC2), that would be great — but it should also be mentioned in the release notes, which may not happen when the ticket remains closed like it is now.

ntrp, 2026-09-19:
Still having issues with this on betafpvf411

```
***** Commencing Build *****
HOST:      foxtrot  (mem 4g, cpus 7, runtime runsc)
VERSION:   2026.6.2
COMMIT:    e0b7bb01b17b21351057e9ead2d1ab39dd44fa16
CONTAINER: 2026.6.X
BOARD:     BETAFPVF411/STM32F411
STARTED:   2026-09-19 00:21:11Z

Want to run this build locally?
> make BETAFPVF411 EXTRA_FLAGS="-D'BUILD_KEY=9b1f1c30a007f10a95cc516de938008c' -D'RELEASE_NAME=2026.6.2' -DCLOUD_BUILD -DUSE_DSHOT -DUSE_NONCOMPLIANT_SMARTAUDIO -DUSE_OSD -DUSE_OSD_SD -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SOFTSERIAL -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"

***** Hardened pipeline log: *****
>>> make BETAFPVF411 -j7 EXTRA_FLAGS="-D'BUILD_KEY=9b1f1c30a007f10a95cc516de938008c' -D'RELEASE_NAME=2026.6.2' -DCLOUD_BUILD -DUSE_DSHOT -DUSE_NONCOMPLIANT_SMARTAUDIO -DUSE_OSD -DUSE_OSD_SD -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SOFTSERIAL -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"
Building target config BETAFPVF411
make --no-print-directory fwo CONFIG=BETAFPVF411
EF HASH -> ./obj/main/STM32F411_BETAFPVF411/.efhash_5b7c01beecfc14eff9f090298a411f26
%% startup_stm32f411xe.s 
%% (size optimised) ./src/platform/common/stm32/config_flash.c 
%% (speed optimised) ./src/platform/common/stm32/system.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (size optimised) ./src/main/drivers/inverter.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/platform/STM32/pwm_output_dshot.c 
%% (optimised) ./src/platform/STM32/adc_stm32f4xx.c 
%% (optimised) ./src/platform/STM32/bus_i2c_stm32f4xx.c 
%% (optimised) ./src/platform/STM32/bus_spi_stdperiph.c 
%% (optimised) ./src/platform/STM32/debug.c 
%% (optimised) ./src/platform/STM32/dma_stm32f4xx.c 
%% (optimised) ./src/platform/STM32/dma_reqmap_mcu.c 
%% (optimised) ./src/platform/STM32/dshot_bitbang_stdperiph.c 
%% (optimised) ./src/platform/STM32/dshot_bitbang.c 
%% (speed optimised) ./src/platform/STM32/exti.c 
%% (optimised) ./src/platform/STM32/io_stm32.c 
%% (optimised) ./src/platform/STM32/light_ws2811strip_stdperiph.c 
%% (optimised) ./src/platform/STM32/persistent.c 
%% (optimised) ./src/platform/STM32/rcc_stm32.c 
%% (optimised) ./src/platform/STM32/sdio_f4xx.c 
%% (optimised) ./src/platform/STM32/serial_uart_stdperiph.c 
%% (optimised) ./src/platform/STM32/serial_uart_stm32f4xx.c 
%% (optimised) ./src/platform/STM32/system_stm32f4xx.c 
%% (optimised) ./src/platform/STM32/timer_stdperiph.c 
%% (optimised) ./src/platform/STM32/transponder_ir_io_stdperiph.c 
%% (optimised) ./src/platform/STM32/timer_stm32f4xx.c 
%% (optimised) ./src/platform/STM32/usbd_msc_desc.c 
%% (optimised) ./src/platform/STM32/camera_control_stm32.c 
%% (optimised) ./src/main/drivers/adc.c 
%% (optimised) ./src/platform/STM32/startup/system_stm32f4xx.c 
%% (size optimised) ./src/main/drivers/serial_escserial.c 
%% (speed optimised) ./src/platform/common/stm32/io_impl.c 
%% (optimised) ./src/platform/common/stm32/adc_impl.c 
%% (size optimised) ./src/platform/common/stm32/bus_i2c_pinconfig.c 
%% (size optimised) ./src/platform/common/stm32/bus_spi_pinconfig.c 
%% (optimised) ./src/platform/common/stm32/serial_uart_hw.c 
%% (size optimised) ./src/platform/common/stm32/pwm_output_beeper.c 
%% (size optimised) ./src/platform/common/stm32/serial_uart_pinconfig.c 
%% (speed optimised) ./src/platform/common/stm32/pwm_output_dshot_shared.c 
%% (optimised) ./src/platform/common/stm32/dshot_dpwm.c 
%% (speed optimised) ./src/platform/common/stm32/dshot_bitbang_shared.c 
%% (optimised) ./src/platform/common/stm32/debug_pin.c 
%% (optimised) ./src/platform/common/stm32/ledstrip_ws2811_stm32.c 
%% (optimised) ./src/platform/common/stm32/expresslrs_driver_hw.c 
%% (optimised) ./src/platform/common/stm32/fault_handlers.c 
%% (optimised) ./src/platform/common/stm32/can_hw.c 
%% (size optimised) ./src/platform/common/stm32/can_pinconfig.c 
%% (optimised) ./src/platform/common/stm32/mco.c 
%% (speed optimised) ./src/platform/common/stm32/bus_spi_hw.c 
%% (optimised) ./src/platform/common/stm32/camera_control.c 
[... 436 lines cut ...]
%% (optimised) ./src/platform/STM32/vcpf4/usb_cdc_hid.c 
%% (size optimised) ./src/platform/STM32/serial_usb_vcp.c 
%% (size optimised) ./lib/main/google/olc/olc.c 
%% (optimised) ./src/main/drivers/usb_io.c 
Linking STM32F411_BETAFPVF411 
Memory region         Used Size  Region Size  %age Used
           FLASH:        7968 B        16 KB     48.63%
    FLASH_CONFIG:           0 B        16 KB      0.00%
          FLASH1:      353104 B       480 KB     71.84%
   SYSTEM_MEMORY:           0 B        29 KB      0.00%
             RAM:       78336 B       128 KB     59.77%
       MEMORY_B1:           0 B          0 B
   text	   data	    bss	    dec	    hex	filename
 355396	   5676	  72248	 433320	  69ca8	./obj/main/betaflight_STM32F411_BETAFPVF411.elf
Creating HEX ./obj/betaflight_2026.6.2_STM32F411_BETAFPVF411.hex 
Building target config BETAFPVF411 succeeded.
>>> Build complete. Artefacts in /source/obj:
  /source/obj/betaflight_2026.6.2_STM32F411_BETAFPVF411.hex (1015690 bytes)

***** Completed Build *****
Build time: 2026-09-19 00:21:11Z, time taken: 20.49
Submitted:  2026-09-19 00:21:09Z, wait time: 1.51
```

Flashed with this firmware but options do not show USE_NONCOMPLIANT_SMARTAUDIO and the vtx says SmartAudio Unknown