betaflight/betaflight #14327: LUXMINIF7 V2 FC Gyro issue

Swerllz, 2025-03-29:
### Describe the bug

ICM42688P fails to initialize in 4.5.2 and 4.5.3. Latest that it recognizes the gryo is 4.3.2.

[# status 4.3.2.txt](https://github.com/user-attachments/files/19522684/status.4.3.2.txt)

[BTFL_cli_4_5_2_LUXMINIF7.txt](https://github.com/user-attachments/files/19522688/BTFL_cli_4_5_2_LUXMINIF7.txt)

[BTFL_cli_4_3_2_LUXMINIF7.txt](https://github.com/user-attachments/files/19522690/BTFL_cli_4_3_2_LUXMINIF7.txt)

### To Reproduce

flash to 4.4.x and newer

### Expected behavior

Flashed firmware on a new build to 4.5.2 and gryo/accel were not responding. Worked through CLI to reset gyro, uploaded cli from 4.3.2, attempted to use 4.4.x, 4.5.x, and even 4.6.0. 4.3.2 was the latest that would detect gyro

### Support ID

```txt
d52cd12d-7a21-47bc-b78f-c39584fbcd61
```

### Flight controller

Lumenier Lux f7 mini V2, from GetFPV

### Other components

ESC: NewBeeDrone Infinity 200 V2 55A
RX: TBS Nano SE
VTX: SpeedyBee TX1600
Cam: Foxeer Predator Nano

haslinghuis, 2025-03-29:
Try adding `ACC_SPI_ICM42688P GYRO_SPI_ICM42688P` in custom build options before flashing.

Swerllz, 2025-03-29:
Thank you! That was it, good to go now!

closed 2025-03-29 by Swerllz