betaflight/betaflight #14328: Disarm caused reboot and GPS rescue to activate

MayanHawk, 2025-04-01:
### Describe the bug

I was out flying today, and had a scary issue. Came in to land, was about 4 inches off the ground, disarmed, and my video went out for a split second **then** I saw ``RX LOSS`` _and_ ``CPU OVERLOAD`` in my OSD as the quad activated GPS rescue mode.  

The quad ascended to about 10 feet, then abruptly disarmed and fell like a brick.  I assume since I was so close to the take-off position, the GPS rescue kicked on, then immediately off resulting in the falling behavior

My sticks were down, radio switch was in the ``disarm`` position, and I never got a telemetry lost call out. 

Board is a Matek H743 and has otherwise been reliable.

### To Reproduce

1. Acquire GPS satellites
2. Arm and fly for more than 3 minutes
3. Arrest speed and descend to ~1 foot above the ground
4. Disarm

### Expected behavior

Quadcopter should disarm, and remain disarmed and safe to approach.

### Support ID

```txt
see attached file ``10in_Endurance_GEPRC.txt``
```

### Flight controller

Matek H743 v1.5 purchsed from GetFPV ~3 years ago

### Other components

- GepRC Mark 4 10" (with braces installed)
- T-Motor/Hobby V3115 900kv motors
- Matek H743 v1.5
- Hobbywing 60A ESC (settings and version attached)
- Foxeer T-Rex Mini
- Foxeer T-Cat3 Mini
- TBS M10 GPS
- Radiomaster ELRS RP4TD receiver
- iFlight Blitz 2.5W VTX
- MenaceRC COB LEDs, with MenaceRC COB Driver

[image]

### How are the different components wired up (including port information)

- C1: Foxeer T-Rex (FC 5v and FC GND)
- C2: Foxeer Cat3 (FC 5v and FC GND)
- TX 4: VTX (FC VBAT (vsw), FC GND, FC video out, Tramp Telemetry)
- RX3: Remapped to PINIO on/off switch for MenaceRC COB driver
- RX/TX 6: ELRS
- RX/TX 2: GPS
- Buzzer to ViFly Finder

### Add any other context about the problem that you think might be relevant here

[10in_Endurance_GEPRC.txt](https://github.com/user-attachments/files/19558422/10in_Endurance_GEPRC.txt)

MayanHawk, 2025-04-01:
Log couldn't upload to the error report, [I've linked it here](https://drive.google.com/file/d/1clOsb8mx2PNpcRUg_l0uJ-xta1dT_Jgw/view?usp=sharing)

MayanHawk, 2025-04-01:
Possibly related: [Issue #12894](https://github.com/betaflight/betaflight/issues/12894#issuecomment-1735338938)

haslinghuis, 2025-04-01:
> Came in to land, was about 4 inches off the ground, disarmed, and my video went out for a split second then I saw RX LOSS and CPU OVERLOAD in my OSD as the quad activated GPS rescue mode.

But disarmed is disarmed - and should not display failsafe [and execute GPS rescue]

Get support ID from CLI tab and post here for reference.

MayanHawk, 2025-04-01:
> But disarmed is disarmed - and should not display failsafe [and execute GPS rescue]
>
> Get support ID from CLI tab and post here for reference.

Whenever I try to generate a support ID, I get a ``Build server failure: HTTP 400``

haslinghuis, 2025-04-01:
@limon could this be caused by OSD stats waiting for IMU being steady

https://github.com/betaflight/betaflight/pull/13637

haslinghuis, 2025-04-01:
@MayanHawk output of 

```
version
status
tasks
resource show all
dump
```

MayanHawk, 2025-04-01:
[10inch_requested data.txt](https://github.com/user-attachments/files/19558945/10inch_requested.data.txt)

Here you go

haslinghuis, 2025-04-01:
Sorry BF does not support KAACK version. @limonspb please have a look

limonspb, 2025-04-04:
@MayanHawk do i understand it correctly that you can reliably reproduce it? Or it happened only one time?
If you can reproduce it, does it still happen with `set stats_min_armed_time_s = -1` (turn off stats saving upon disarm)?

@haslinghuis, PR https://github.com/betaflight/betaflight/pull/13637 is in 4.6, not 4.5...

Weirdly the log just stops abruptly...

I think it's a good guess that it's related to EEPROM saving upon disarm. In this case KAACK is to blame for only turning this stats saving by default, and it can happen with regular BF + `set stats_min_armed_time_s = something > 0`...

MayanHawk, 2025-04-04:
> [@MayanHawk](https://github.com/MayanHawk) do i understand it correctly that you can reliably reproduce it? Or it happened only one time? If you can reproduce it, does it still happen with `set stats_min_armed_time_s = -1` (turn off stats saving upon disarm)?
> 

I was only able to get the behavior once. I tried five or six times with props off but never got the same result. I went as far as arming and carrying the quad up and down the street to get to the same approximate flight distance; disarm worked as expected.

> Interesting detail that video went out for a split second. I see you using two cameras on a switch between them. When video went out, did you see static, or grey/black? Did you see OSD while video was out? If you do actually failsafe (turn off radio), does your video turn off? DVR would be very useful, but i understand you probably don't record every flight...
> 

[Here is a link to the DVR.](https://drive.google.com/file/d/1XV8NiOdViz_egFNka2owgWiRJjgBhYee/view?usp=sharing) My goggles weren't recording, but my VO's monitor caught the whole flight. The behavior begins at ~6:03, here's a quick breakdown:

- Hovering ~4" above the ground at <22% throttle. As I started settling into the grass, I decreased throttle to 0% and flipped the arm switch.
- Disarming immediately results in the video going black, then grey.
- About 1 second later, the OSD elements repopulate the screen. All values are the same as before video blacked out. The message ``RX LOSS`` is visible.
- About 1.5 seconds from video blackout, video restores. Quad is in an uncontrolled climb. OSD indicates throttle is at 0%.
- About 0.5 seconds later, the video cuts out to grey but OSD elements remain.
- About 1 second later, the quad displays ``FAILSAFE`` and begins to fall from ~70 feet
- About 0.7 seconds later, while in free fall, the OSD displays ``CPU OVERLOAD``

> Weirdly the log just stops abruptly like a power off...
> 

I noticed the log cuts in and out briefly at 3m2s. Is this indicative of early CPU overload?

> I think it's a good guess that it might be related to long EEPROM saving on H7 upon disarm. In this case KAACK is to blame for only turning ON this stats saving by default, and then it can happen with regular BF + `set stats_min_armed_time_s = something > 0`... But the fact that video went out puzzling me... Electrical glitch upon disarm?
> 

Every H7 I own does a reboot when I disarm regardless of stats being on or off. Since this incident, I've been doing a half flip right before disarming so I land upside down.

> [@MayanHawk](https://github.com/MayanHawk) i'd recommend to turn off stats saving, and carefully see how that goes... I don't have H7 FC to check myself, might get one to just check this.

I already flashed to stable release 4.5.2 and have been waiting for weather to improve before testing with stats disabled. I don't know if I can get the behavior to repeat though.

limonspb, 2025-04-04:
> I noticed the log cuts in and out briefly at 3m2s. Is this indicative of early CPU overload?
i think that log brief interruption happens with SD cards occasionally. Seen it many times with all processors.

> Every H7 I own does a reboot when I disarm regardless of stats being on or off. Since this incident, I've been doing a half flip right before disarming so I land upside down.
Is it actually FC reboot, or just ESC initialization tones? So when this "reboot" happens, do you see OSD re-initialization with Betaflight logo and stuff?

In my understanding the video loss is somthing FC can't do no matter if it freezes, glitches etc. Unless you have a video ON/OFF on a switch. Because you have VTX powered from vbat. If FC can't cause video loss, why it happened then?

What if you do the following:
1. Flash the version you had, and set it up
2. props off, plug the battery, make video and OSD to show on screen.
3. Connect with Betaflight configurator over USB.
4. Change some parameter somewhere on PID tab
5. Hit save and reboot making FC to reboot.
6. What will happen with the video?

limonspb, 2025-04-04:
@MayanHawk interesting, so it's a full FC reboot upon disarming. Does it happen with regular 4.5.2 release and 4.6, if you willing to try?

MayanHawk, 2025-04-04:
@limonspb 

Regular 4.5.2
- **Adjusted PID slider clicked save** - RESULT: OSD flashed ``RX Loss`` once. Only got this result once out of five times.
- **Loaded the ``ExpressLRS 500hz`` preset, then pressed ``Save and Reboot``** - RESULT: Video went to static; boot tones heard; video returned with the betaflight boot screen. OSD loaded normally.

Dev version 4.6
- **Adjusted PID slider clicked save** - RESULT: no abnormal behavior
- **Loaded the ``ExpressLRS 500hz`` preset, then pressed ``Save and Reboot``** - RESULT: Video went to static; boot tones heard; video returned with the betaflight boot screen. OSD loaded normally.

haslinghuis, 2025-04-04:
Boot tones on `Save and reboot` would be expected behavior. Same with changing PID settings.

IMO the issue is a reboot on disarm.

limonspb, 2025-04-04:
Ok so we at least ruled out KACK, that it's not related and issue happens with official 4.5 and 4.6...
Time for heavy lifters like Mr Steve?

MayanHawk, 2025-04-04:
Will test Monday, I've got a very busy weekend ;)

MayanHawk, 2025-04-07:
> [@MayanHawk](https://github.com/MayanHawk) can you plz check official 4.5.2 and 4.6, does it actually reboot upon disarming? (Without connection by USB, just battery) With stats ON and OFF.

**4.6.0**
- Stats disabled, armed for 10s, disarmed: No video loss, and post flight stats displayed (``used mah`` element was updating while the tones were playing).
- Stats enabled, armed for 10s, disarmed: Heard 3x ESC initialization tones and 2x FC tones. No video loss, and post flight stats displayed (``used mah`` element was updating while the tones were playing).

**Release 4.5.2**
- Stats disabled, armed for 10s, disarmed: No video loss, and post flight stats displayed (``used mah`` element was updating while the tones were playing).
- Stats enabled, armed for 10s, disarmed: Heard 3x ESC initialization tones and 2x FC tones. No video loss, and post flight stats displayed (``used mah`` element was updating while the tones were playing).

**KAACK v17 4.5.2**
- Stats disabled, armed for 10s, disarmed: No video loss, and post flight stats displayed (``used mah`` element was updating while the tones were playing).
- Stats enabled, armed for 10s, disarmed: Prior to arming, when plugging in battery, all five initialization tones sounded. After arming, Heard 3x ESC initialization tones and 2x FC tones. No video loss, and post flight stats displayed (``used mah`` element was updating while the tones were playing).

limonspb, 2025-04-08:
@MayanHawk  so in any of the cases the FC is not actually rebooting? Just ESC initialization tones... But V15 were rebooting with Betaflight logo?

MayanHawk, 2025-04-08:
@limonspb You're welcome!

You're making me doubt myself now. I'm convinced I saw the Betaflight logo in my goggles, but I never saw it in the dvr I shared with you.

I was was under the impression that the initialization tones indicated a full reboot, but I haven't actually seen the ``full Betaflight logo and stuff``.

closed 2025-05-16 by github-actions[bot]