betaflight/betaflight #14330: The GPS satellite count disappears during full throttle, showing a “Rescue N/A” notification.

djek88, 2025-04-02:
### Describe the bug

During full throttle in a horizontal position, GPS satellites count is 0 for one second, and a “Rescue N/A” notification is shown.
The same behavior occurs during a long vertical dive and a fast stop before hitting the ground.

### To Reproduce

Hover in a horizontal position and apply full throttle to make the drone ascend quickly.
Dive vertically and stop suddenly just before reaching the ground.
Pls check the video: https://drive.google.com/file/d/1IyX2Yglfe9JE3T85jDkqe9lwWWxJjUU9/view?usp=sharing

### Expected behavior

GPS works properly during full throttle, no "Rescue N/A" notification is shown.

### Support ID

```txt
4.5.2 where is gps bug is present:
Id: "5557b98b-6eeb-4673-a681-fe199f0eb982"

Commit ([4.5.2] Fix SA for holybro on AT32F4 (#14246) 5f6e00d904529a99c5234ddb2ce35a88aaad15f9) where the bug is not present:
Id: "03470fa9-a137-4d15-9460-c6713af32bb5"
```

### Flight controller

Holybro Kakute H7 v1.3 (MPU6000)

mituritsyn, 2025-04-02:
@haslinghuis  The issue appears in #14299  merge commit and does not reproduce in the commit before it.

djek88, 2025-04-04:
yes with AIRBORNE_4G setting everything works fine

0crap, 2025-04-04:
Funny, I have exactly the same issue.
First flight this season.
I updated firmware from where I left it last season (4.5.0) to 4.5.2.
I suspect also #14299 so I flashed back to 4.5.1
Will post my findings when I fly again.

haslinghuis, 2025-04-04:
@0crap still can use 4.5.2 just change `gps_ublox_flight_model` to `AIRBORNE_4G`.

Rather surprised why this issue exists.

0crap, 2025-04-04:
Awesome, I did just that. Will report back.
My guess of why, is that the `UBLOX_MODEL_AIRBORNE_1G` setting is more susceptible for interference?
Happens only on a full throttle punch, just like the OP reports.

Using a [WS-M181 GPS](https://caddxfpv.com/products/walksnail-ws-m181gps).

haslinghuis, 2025-04-04:
See also 
- https://github.com/betaflight/betaflight/pull/12587
- https://github.com/betaflight/betaflight/pull/12659#issuecomment-1509049645

haslinghuis, 2025-04-04:
@ZzyzxTek still around ?

ZzyzxTek, 2025-04-05:
> [@ZzyzxTek](https://github.com/ZzyzxTek) still around ?

Yeah, just not actively doing anything...

ZzyzxTek, 2025-04-05:
So, back when we did the original flight model changes and testing, we landed on 4G as the default for this very reason. The thinking being we wanted a default that would work for everyone, even if not ideal.

When I saw the default changed to 1G, I thought, OK but that might not work when people that fly more extreme with a GPS.

The flight modes of 1G, 2G, 4G really have nothing to do with G forces at all, that's just the way UBLOX decided to label them. They control the amount and type of filtering the module applies.

1G has a lot less filtering, and seemed generally fine for long range cruising. 1G would theoretically have more accurate position updates. But it would be more succeptible to rapid or jerky position changes. I didn't know it might totally stop reporting a position if the amount of change exceeded its threshold, but apparently it might.

4G has a lot more filtering. Opposite plus/minus, more accurate at high speeds or rapid position changes. Less accurate at smooth flight.

If 1G is the desired default to accommodate the probably more common long range use case with less dynamic flight, a user like the OP should be advised to try 2G then 4G if they fly more dynamic as described.

closed 2025-04-06 by haslinghuis

denis-sher, 2025-04-08:
can it be set /codeed for acro and angle to be 4G and position hold and altitude hold RTH to be 1G ?

PR #14334 (2025-04-04): [4.5.3] Revert gps flight model
- reverts https://github.com/betaflight/betaflight/commit/024f8e13d4e642eb6a380308685b9ea3aa3ef1a2
- fixes #14330
PR #14333 (2025-04-06): Revert GPS flight model change
- reverts https://github.com/betaflight/betaflight/commit/92c59091cee24c48d0cf142b8ee0b0b5dfc07c2e
- fixes #14330