betaflight/betaflight #14351: "osd_esc_rpm_alarm" option should use motor RPM reported by DShot

dmak, 2025-04-18:
### Is your feature request related to a problem? Please describe

* Set `osd_esc_rpm_alarm=1000`
* Start flying the drone. OSD warning "ESCRRRR" is permanently on.

### Describe the solution you'd like

`osd_esc_rpm_alarm` should use RPM values reported by DShot (if it is configured and working).

### Describe alternatives you've considered

-

haslinghuis, 2025-04-18:
Same behavior with `ESC_SENSOR` disabled ?

lasa01, 2025-04-24:
The RPM alarm does work with bidirectional dshot, but only if dshot_edt is enabled in the cli.

Though this doesn't seem right, since the RPM information is available even without edt. The code confirms that all the alarm updates are skipped (rpm, temp and current), for motors with no extended telemetry available: https://github.com/betaflight/betaflight/blob/5a6d13c1a6207021ce77e4e53d0ba2fe689c4516/src/main/osd/osd_warnings.c#L339-L342

It would make sense to allow displaying the RPM alarm even without edt, since the RPM data is available anyway.

haslinghuis, 2025-04-24:
Please test #14361

lasa01, 2025-04-24:
Tested this on a whoop.

[image]

The alarm text seems to be always visible, even if there is no alarm to show. Didn't have this problem with 4.5. The RPM alarm does work now even without dshot_edt (and without ESC_SENSOR), the "ESC 1 2 3 4" text is just shown all the time. When RPM alarm is active for motor 4 for example, this shows "ESC 1 2 3 4R".

haslinghuis, 2025-04-24:
PR has been updated. Please test again

lasa01, 2025-04-24:
I think this is incorrect: https://github.com/betaflight/betaflight/pull/14361/commits/057258959fcbd9f707fa1f7695b53e4b80f735e3
We want to have RPM alarm when RPM is too low to detect motor stalls.

The problem is probably that this code gets skipped with the continue you added earlier, leading to the motor numbers staying in the warning, even though the RPM warning is not active:
https://github.com/betaflight/betaflight/blob/057258959fcbd9f707fa1f7695b53e4b80f735e3/src/main/osd/osd_warnings.c#L369-L371

haslinghuis, 2025-04-24:
@lasa01 Thanks. Updated again to include above code.

lasa01, 2025-04-24:
Almost works, now it just detects too high RPM, when it should detect too low RPM. Should probably revert the >= change.

haslinghuis, 2025-04-24:
@lasa01 done

lasa01, 2025-04-25:
Tested and looks good, works as expected both dshot_edt off and dshot_edt on.

dmak, 2025-04-28:
> Same behavior with `ESC_SENSOR` disabled ?

Thanks, worked for me with `feature ESC_SENSOR` disabled as well.

By the way, is `osd_esc_rpm_alarm` documented? It means "minimal RPM"?

Linking also issue #12905.

closed 2025-05-15 by blckmn

haslinghuis, 2025-05-19:
@dmak added back port for 4.5.3 in [#14394](https://github.com/betaflight/betaflight/pull/14394)

PR #14361 (2025-05-15): Fix osd_esc_rpm_alarm
- fixes #14351

## Summary by CodeRabbit

- **Bug Fixes**
	- Improved warning display logic to ensure temperature and current alarms are only shown for motors with available telemetry data. All motors are now processed for warnings, but checks are applied appropriately based on telemetry presence.