betaflight/betaflight #14354: Failsafe procedure "GPS Rescue" is back to "Fall" after restart

MeeJay, 2025-04-20:
### Describe the bug

When selecting "GPU Rescue" as Failsafe Procedure, the drone always go back to "Fall" procedure after clicking on "Save & Restart" even with a working GPS and "Phase 1" correctly set-up.

### To Reproduce

Go to "Failsafe", use the "GPU Rescue" option and click on "Save & Restart".
Go back to Failsafe after the drone is back and the option will be back to "Fall".

### Expected behavior

The drone should be in GPS Rescue as failsafe procedure.

### Support ID

```txt
345b4c89-c0ba-4834-942c-83dbf5a248d4
```

### Flight controller

iFlight Blitz F435, bought from Banggood with the Nazgul DC5 Eco

### Other components

DJI O4 Air Unit Pro with DJI controller

### How are the different components wired up (including port information)

GPS on UART4, 115200 Baudrate as set-up by iFlight & recommended by them.

### Add any other context about the problem that you think might be relevant here

iFlight Nazgul DC5 Eco with iFlight Blitz F435 Flight Controller.
I've tried with Betaflight 10.10.0 on Windows 11 and using the APK on my phone.
I've also tried with the web configurator of Betaflight on both.

haslinghuis, 2025-04-20:
```bash
GPS: NOT CONNECTED, UART4 38400 (set to 115200), NOT CONFIGURED, version =  unknown
```

- As GPS is not detected it won't accept the setting.
- For further support please checkout our [Discord](https://discord.com/invite/gV4XSq3fus)

closed 2025-04-20 by haslinghuis

MeeJay, 2025-04-21:
Hello,

Thank for your answer.

I was indoor and without battery pack this time, but i've tried on flight site yesterday, with battery and GPS fix, and it did not work.
I'll be outdoor later today, I'll try again to set it to GPS Rescue, if it still not working should I open a new ticket or update this one ?

haslinghuis, 2025-04-21:
For support please checkout our [Discord](https://discord.com/invite/gV4XSq3fus) first.