betaflight/betaflight #14355: Building under msys2 (ucrt64) results in a link failure

hydra, 2025-04-23:
### Describe the bug

Attempting to build firmware results in a build failure:

```
Linking APM32F405
make: the '-j' option requires a positive integer argument
```

### To Reproduce

1) install msys2 (ucrt64)
2) install build dependencies
3) run `make`
4) observe build failure.
5) same behavior when building a single target, e.g. `make SPRACINGH7EXTREME`

### Expected behavior

Successful build.

### Support ID

```txt
N/A
```

### Flight controller

N/A

### Other components

N/A

### Add any other context about the problem that you think might be relevant here

this seems to be caused by the following `Makefile` source:

```
# some targets use parallel build by default
# MAKEFLAGS is valid only inside target, do not use this at parse phase
DEFAULT_PARALLEL_JOBS 	:=    # all jobs in parallel (for backward compatibility)
MAKE_PARALLEL 		     = $(if $(filter -j%, $(MAKEFLAGS)),$(EMPTY),-j$(DEFAULT_PARALLEL_JOBS))
```

running this works:

```
make SPRACINGH7EXTREME DEFAULT_PARALLEL_JOBS=1
```

Suggest detecting msys2 and providing a sensible default.

```
$ make -v
GNU Make 4.4.1
Built for x86_64-pc-msys
$ uname
MINGW64_NT-10.0-22631
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc.exe (Arm GNU Toolchain 13.3.Rel1 (Build arm-13.24)) 13.3.1 20240614
```

nerdCopter, 2025-04-23:
@ledvinap

ledvinap, 2025-04-24:
Can you test if 'nproc' is available in your build environment?

IIRC you can pass '-j 4' to make to override default and avoid this problem