betaflight/betaflight #14357: Dual Gyro not working correctly (SPRacingH7EXTREME)

hydra, 2025-04-23:
### Describe the bug

BF 4.5.0:

https://github.com/user-attachments/assets/087123a7-f79e-4290-88d2-d5d244069404

```
# version
# Betaflight / STM32H750 (SP7E) 4.5.0 Jan  6 2024 / 10:34:49 (2caed9b60d) MSP API: 1.46
# board: manufacturer_id: SPRO, board_name: SPRACINGH7EXTREME

# status
MCU H750 Clock=400MHz, Vref=3.29V, Core temp=57degC
Stack size: 2048, Stack address: 0x20020000
Configuration: CONFIGURED, size: 3865, max available: 131072
Devices detected: SPI:2, I2C:1
Gyros detected: gyro 1, gyro 2 locked dma
GYRO=ICM20602, ACC=ICM20602, BARO=BMP388
OSD: MAX7456 (30 x 13)
System Uptime: 150 seconds, Current Time: 2025-04-23T13:20:46.862+00:00
CPU:20%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 9
SD card: Startup failed
FLASH: JEDEC ID=0x00efaa21 128M
Arming disable flags: RXLOSS CLI MSP

# get gyro_1_align
gyro_1_align_roll = 0
Allowed range: -3600 - 3600

gyro_1_align_pitch = 0
Allowed range: -3600 - 3600

gyro_1_align_yaw = 1800
Allowed range: -3600 - 3600

# get gyro_2_align
gyro_2_align_roll = 0
Allowed range: -3600 - 3600

gyro_2_align_pitch = 0
Allowed range: -3600 - 3600

gyro_2_align_yaw = 0
Allowed range: -3600 - 3600

# get sensor_align
gyro_1_sensor_align = CW180
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP, CUSTOM

gyro_2_sensor_align = CUSTOM
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP, CUSTOM

# get debug_mode
debug_mode = DUAL_GYRO_DIFF
```

BF 4.2.1

https://github.com/user-attachments/assets/a86edeba-162c-44dd-bc1b-3ed076a43e38

```
# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.2.11 Nov  9 2021 / 20:26:14 (948ba6339) MSP API: 1.43

# status
MCU H750 Clock=400MHz, Vref=3.30V, Core temp=61degC
Stack size: 2048, Stack address: 0x20020000
Config size: 3867, Max available config: 131072
Gyros detected: gyro 1, gyro 2
GYRO=ICM20602, ACC=ICM20602, BARO=BMP388
OSD: MAX7456
System Uptime: 83 seconds, Current Time: 2025-04-23T13:28:45.963+00:00
CPU:1%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 226 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 3
SD card: None inserted
Arming disable flags: CLI MSP MOTOR_PROTO

# get gyro_1_align
gyro_1_align_roll = 0
Allowed range: -3600 - 3600

gyro_1_align_pitch = 0
Allowed range: -3600 - 3600

gyro_1_align_yaw = 1800
Allowed range: -3600 - 3600

# get gyro_2_align
gyro_2_align_roll = 0
Allowed range: -3600 - 3600

gyro_2_align_pitch = 0
Allowed range: -3600 - 3600

gyro_2_align_yaw = 2250
Allowed range: -3600 - 3600

# get sensor_align
gyro_1_sensor_align = CW180
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP, CUSTOM

gyro_2_sensor_align = CUSTOM
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP, CUSTOM

# get debug_mode
debug_mode = DUAL_GYRO_DIFF
```

each sensor works find individually.

### To Reproduce

1) flash 4.5.0 reset to defaults, observe gyro doe cause the correct movements in the 'setup' tab (it's jerky) enable dual_gyro_diff debug logging, observe there is a massive (100+) difference on x/y/z axis when moving the FC.
2) flash 4.2.1, reset to defaults, observe gyro is correct in 'setup' tab, enable dual_gyro_diff debug logging, observe there is a < 10 difference on x/y/z axis when moving the FC.

### Expected behavior

Working gyros by default on 4.5.0.

### Support ID

```txt
N/A
```

### Flight controller

SPRacingH7EXTREME

### Other components

None

### How are the different components wired up (including port information)

N/A

hydra, 2025-04-23:
Not sure if this is related.

https://github.com/betaflight/config/pull/645

Seems to be working in `master`. (`eb7814d1970dc791a5d821cb40c9e07c7e00acc0`)

https://github.com/user-attachments/assets/79effbea-ac67-4ead-a0c2-919b92eee622

```
# version
# Betaflight / STM32H750 (SP7E) 4.6.0 Apr 23 2025 / 15:00:42 (eb7814d19) MSP API: 1.47
# config rev: ad70d2a
# board: manufacturer_id: SPRO, board_name: SPRACINGH7EXTREME

# status
MCU STM32H750 Clock=400MHz, Vref=3.31V, Core temp=65degC
Stack size: 2048, Stack address: 0x20020000
Configuration: CONFIGURED, size: 4479, max available: 65536
Devices detected: SPI:2, I2C:1
Gyros detected: gyro 1, gyro 2 locked dma
GYRO=ICM20602, ACC=ICM20602, BARO=BMP388
OSD: MSP (53 x 20)
System Uptime: 314 seconds, Current Time: 2025-04-23T13:41:05.767+00:00
CPU:18%, cycle time: 125, GYRO rate: 8000, RX rate: 0, System rate: 9
Voltage: 0 * 0.01V (0S battery - INIT)
I2C Errors: 9
SD card: None inserted
FLASH: JEDEC ID=0x00efaa21 128M
Arming disable flags: RXLOSS CLI MSP

# get gyro_1_align
gyro_1_align_roll = 0
Allowed range: -3600 - 3600

gyro_1_align_pitch = 0
Allowed range: -3600 - 3600

gyro_1_align_yaw = 1800
Allowed range: -3600 - 3600

# get gyro_2_align
gyro_2_align_roll = 0
Allowed range: -3600 - 3600

gyro_2_align_pitch = 0
Allowed range: -3600 - 3600

gyro_2_align_yaw = 2250
Allowed range: -3600 - 3600

# get sensor_align
gyro_1_sensor_align = CW180
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP, CUSTOM

gyro_2_sensor_align = CUSTOM
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP, CUSTOM

# get debug_mode = DUAL_GYRO_DIFF
```

It seems like 4.5.0 has an incorrect setting: `gyro_2_align_yaw = 0` which should be `gyro_2_align_yaw = 2250` (as per 4.6.0 and 4.2.1)

hydra, 2025-04-23:
Seems to be OK in 4.5.2, so probably nothing to do, but would be nice to know when it broke, when it was discovered and when it was fixed (issue/PR links and commit hashes would be great).

nerdCopter, 2025-04-23:
4.5.0 should not build any different than 4.5.2 🤔 , unless 4.5.0 code is not honoring the config.h

target references (for easy click):
- https://github.com/betaflight/config/blob/ad70d2a88ff301173fc9e2b57c298b472a978848/configs/SPRACINGH7EXTREME/config.h#L247-L248
- https://github.com/betaflight/unified-targets/blob/d20eea23920b0cb9b49cd95b29cb8c860318ac3c/configs/default/SPRO-SPRACINGH7EXTREME.config#L214C1-L217C28

haslinghuis, 2025-04-23:
Fixed in 4.5.2 and 4.6.0

Rational behind the change was in https://github.com/betaflight/betaflight/commit/c188a03beb351ccf3a7efaa0d0007b5f4d059a0b

closed 2025-04-23 by haslinghuis

hydra, 2025-04-23:
Would also be nice to know exactly which commit broke it, and why the release process didn't catch this issue as it apparently affected many targets.