betaflight/betaflight #14364: OSD RPM alarm activated instantly after arming in crash flip mode

lasa01, 2025-04-25:
### Describe the bug

When the OSD RPM alarm is enabled and crash flip mode activated, and the quad is armed, the RPM alarm is instantly activated for all motors. The FC is not even trying to spin the motors until the user gives some stick inputs, so their RPM is expected to be zero and the alarm should only be considered when the motors are actually trying to spin.

RPM alarm would be useful in crash flip mode, since if a motor is stuck, the alarm could be used to stop trying to crash flip to prevent damage to the motor or esc. However, currently the alarm is always active for the motors that are not even trying to spin, making it difficult for the user to see when motors are actually stuck in crash flip mode.

Example of activating crash flip mode (and spinning the motors one side at a time)
[image]

### To Reproduce

- Enable the OSD RPM alarm and set up crash flip mode
- Make sure Crash flip mode warning is disabled in OSD, since this is displayed on top of the RPM alarm
- Power on the quad, activate crash flip mode and arm, without giving any stick inputs
- See that the RPM alarm is instantly activated for all motors, even though the quad is not even trying to spin the motors until the sticks are moved

### Expected behavior

- In crash flip mode (or why not in all modes?), RPM alarm is only activated for those motors that are given commands to spin by the FC and are not reaching the RPM threshold. I.e., the alarm is not activated for the motors the FC is not even trying to spin.
- Additionally, IMO, the RPM alarm warning in the OSD should have higher priority than the Crash flip mode warning, which is currently displayed on top of the RPM alarm if it is enabled, hiding the RPM alarm.

### Support ID

```txt
37553a2e-a112-43e7-9087-faceedd99c5f
```

### Flight controller

BetaFPV Air G4 5in1

### Add any other context about the problem that you think might be relevant here

- Had same issue in 4.5.0 and with other quads

lasa01, 2025-06-02:
Just tested the latest dev build, unfortunately still have the same issue

nerdCopter, 2025-07-08:
@coderabbitAI, propose a fix.

lasa01, 2025-08-01:
> [@lasa01](https://github.com/lasa01), please test [#14561](https://github.com/betaflight/betaflight/pull/14561)

Seems like you're having some problems with getting the AI to act on the suggested changes on that PR, maybe I will wait until those are resolved :D

haslinghuis, 2025-08-10:
@lasa01 did you test it ?

lasa01, 2025-08-16:
@haslinghuis tested this and left a comment on the PR. Looks pretty good already, but I wrote one more suggestion for improvement.

PR #14566 (2025-08-17): Fix RPM alarm check
- continue from #14561
- fixes #14364
- should we just stick with first commit only :)
- unified warnings for esc_sensor and dshot into single function
- output should be like ESC 1RT 3C (only show motor number on alarm)
- ~~adds V for EDT voltage alarms~~

## Summary by CodeRabbit

- New Features
  - OSD now displays ESC warnings using both ESC sensors and DShot telemetry.
  - Per-motor alarm indicators added for RPM, current, and temperature.
  - Warning messages are centered when shown and use blinking for visibility.

- Refactor
  - Unified ESC warning generation for co [... cut]