betaflight/betaflight #14368: Motor 3 Issue: Resource Mapping FLYWOOF745 (GOKU F745 HEX FC) Betaflight 4.6 [Latest] Dev

drclawscat, 2025-04-26:
### Describe the bug

I have an Explorer V2 that came with a Factory F745hex board installed and using FLYWOO745 fw.
I've gone down this upgrade path 4.2 --> 4.28 --> 4.3.2  --> 4.4.1 --> 4.6 Dev		

After upgrading from 4.4.1 --> 4.6 motor 3 doesn't show Dshot telemetry and won't spin up. Upon inspection of resources the motor resource mappings are different from factory.  I recall I had similar issues with 4.3.2/4/.4.1 upgrades but managed to remap resources before.  With 4.6 I have USB disconnects before completing mappings.

4.6

[image]

[image]

resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 C09
resource MOTOR 6 A03
resource MOTOR 7 B04
resource MOTOR 8 B05

CORRECT mappings  (From Flywoo Support):
resource MOTOR 1 B01
resource MOTOR 2 C09
resource MOTOR 3 B00
resource MOTOR 4 E11
resource MOTOR 5 E09
resource MOTOR 6 A03
resource MOTOR 7 B04
resource MOTOR 8 B05

Trying all combinations of resource re-mappings to get pins assigned correctly always ends up with the USB disconnected and requiring a power to bootloader and reflash to recover as I can't access BF.  Also after a fresh flash, switching configuration from Quad X to Hex + results in USB disconnect requiring a reflash as well.  Here is the full resource dump from a fresh flash:

resource show all
Currently active IO resource assignments:
(reboot to update)
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_SDI 1
A07: SPI_SDO 1
A08: USB_DETECT
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_SDI 2
B15: SPI_SDO 2
C00: PINIO 1
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_SDI 4
E06: SPI_SDO 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
CH2 : DSHOT_BITBANG 2
CH4 : DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_SDI 4
DMA2 Stream 1: SPI_SDO 4
DMA2 Stream 2: SPI_SDI 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_SDO 1
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 5

Closest I've gotten is this order in the CLI after dozens of combinations:

resource MOTOR 1 none
resource MOTOR 2 none
resource MOTOR 3 none
resource MOTOR 4 none
resource MOTOR 5 none
resource MOTOR 6 none
resource MOTOR 7 none
resource MOTOR 8 none
Save
resource MOTOR 3 B00
Save
resource MOTOR 4 E11
Save
resource MOTOR 2 C09
Save
resource MOTOR 1 B01
Save (loose USB connection here and need to reflash)

### To Reproduce

Flash 4.6 [Latest] target FLYWOOF745 to LR Explorer V2 with Factory F745 Hex layout board.

### Expected behavior

all motors spin up properly and mapped correctly

### Support ID

```txt
ed55c5db-7907-4624-9a8b-fd8959a6366f
```

### Flight controller

Goku F745HEX, bought from Flywoo came with a Explorer LR V2

### Other components

Crossfire Nano RX, Caddx Vista, Flywoo Bluetooth Nano V2, Flywoo Nano GM10 V3 w/ compass, Vifly Finder

### How are the different components wired up (including port information)

Pin connection between 13A ESC Stack and FC, UART1 Crossfire RX, UART2 GPS, UART3 Caddx Vista, UART5 Bluetooth, UART7 ESC Telemetry (reserved in FC)

nerdCopter, 2025-06-03:
@drclawscat, you need to use the target `FLYWOOF745HEX`

```
configs/FLYWOOF745HEX/config.h:41:#define MOTOR1_PIN           PB1
configs/FLYWOOF745HEX/config.h-42-#define MOTOR2_PIN           PC9
configs/FLYWOOF745HEX/config.h-43-#define MOTOR3_PIN           PB0
configs/FLYWOOF745HEX/config.h-44-#define MOTOR4_PIN           PE11
configs/FLYWOOF745HEX/config.h-45-#define MOTOR5_PIN           PE9
configs/FLYWOOF745HEX/config.h-46-#define MOTOR6_PIN           PA3
configs/FLYWOOF745HEX/config.h-47-#define MOTOR7_PIN           PB4
configs/FLYWOOF745HEX/config.h-48-#define MOTOR8_PIN           PB5
```

closed 2025-06-03 by nerdCopter

drclawscat, 2025-06-03:
Hi @nerdCopter , thanks for the reply.   I've loaded that Hex version (it's currently loaded) but I can't access BF at all with it on and can only get into the bootloader now using the ImpluseRC driver fix. It took different laptop attempts to load it since it was giving error messages before erasing on one machine.

So the HEX Version on 4.6 seems to be DOA for some reason on my board.   I haven't tried flashing 4.5 latest HEX as I was upgrading to 4.6 to access Postion and Altitude hold.

Any other ideas?

nerdCopter, 2025-06-04:
interesting.  if the non "HEX" FC works with motor reassignment, then i suppose use it.  maybe there were different hardware versions that were not advised of.  try [Betaflight discord](https://discord.betaflight.com/) for "support", i believe you could ping @brandonsbakedbeans as i think he may have this FC (or maybe it was some other "hex").  

regardless Impulse Driver Fixer is a common (and reoccurring) need on windows, as well as Zadig. see: https://hackmd.io/@nerdCopter/rJv5TUrQ2

drclawscat, 2025-06-05:
> interesting. if the non "HEX" FC works with motor reassignment, then I suppose use it. maybe there were different hardware versions that were not advised of. try [Betaflight [...]

yes, I've gone from 4.2 up to 4.41 (4 upgrades) using the non "HEX" but have had to remap motor pins (one motor is always unmapped and the order is wrong on the default target).  On 4.6 remapping pins at some point causes it to brick and requires reflashing to get in.  So right now it seems impossible to get one of the motors working on the non "Hex". 4.6 Hex just won't work at all and no access to Betaflight.

I can try the discord and will reach out to @brandonsBakedBeans, thanks!

Denethor85, 2025-09-16:
Same FC, same problem here. I bought a used Flywoo Firefly Hex Nano HD with this same FC and it arrived in version 4.2. I updated to version 4.5.2, but as soon as I installed the update, I couldn't connect to the USB port. The only thing that helped was the forced DFU mode with the boot button and the firmware update. Version 4.4 seems to work fine, so the problem probably arises from 4.5 onwards.