betaflight/betaflight #14369: ExpressLRS SPI aux channel decode error

Fourflies, 2025-04-26:
### Describe the bug

In Betaflight, Aux channels get decoded to incorrect values. It looks like an integer overflow or an MSB getting chopped in the channel decoder. When the transmitter sends in the range 1000-1500 µs it gets decoded in Betaflight to a range of 1000-2000 µs. Just as the transmitter goes over 1500 µs, the Betaflight decode jumps back to 1000 µs and then the transmitter's 1500-2000 µs range gets decoded as 1000-2000 µs in Betaflight. Only occurs on the Aux channels. Channels 1-4 are unaffected.

### To Reproduce

Specific to ExpressLRS SPI in Betaflight. Unable to reproduce with a serial ELRS receiver. Confirmed in Betaflight 4.5.2, 4.5.1, and 4.4.3.

Narrowed down to an interaction between Wide switch mode and telemetry. In Hybrid switch mode, channel decodes work correctly regardless of telemetry. In Wide switch mode, if telemetry is disabled in the TX module, the Aux channel decode bug occurs. Enabling any telemetry ratio in the TX module fixes the issue.

### Expected behavior

Correct aux channel decodes.

### Support ID

```txt
a50fac03-f2e2-4bd6-bc00-451a42f81461
```

### Flight controller

HDZero AIO5

### Add any other context about the problem that you think might be relevant here

Wild guess, but perhaps the issue is in this function: https://github.com/betaflight/betaflight/blob/master/src/main/rx/expresslrs.c#L271

haslinghuis, 2025-04-26:
SPI ELRS code is not being maintained by Betaflight.

haslinghuis, 2025-04-26:
@phobos- or @CapnBry may have a look at it.

closed 2025-04-26 by haslinghuis

Fourflies, 2025-04-26:
@haslinghuis Where should this ticket be logged? ExpressLRS don't want it, and it seems Betaflight don't want it either. Yet the code lives in Betaflight.

deadbytefpv, 2025-04-27:
> [@haslinghuis](https://github.com/haslinghuis) Where should this ticket be logged? ExpressLRS don't want it, and it seems Betaflight don't want it either. Yet the code lives in Betaflight.

It's not that the ExpressLRS team don't want it. It's just that the SPI code is within the Betaflight codebase.

ledvinap, 2025-04-27:
There was issue with buggy ELRS transmiter recently, this may be duplicate

Fourflies, 2025-05-03:
@CapnBry I tried to reproduce the issue with a 4.6.0-dev build. I can confirm it is fixed in master. Thank you for chiming in and for the fix!

@deadbytefpv Thank you too for your assistance.