betaflight/betaflight #14370: Vibration figure in OSD

AirCombat, 2025-04-27:
### Is your feature request related to a problem? Please describe

As a FPV pilot I would like to know how aggressively the quad is vibrating so that I can understand its health and any abnormal behaviour.

### Describe the solution you'd like

In aviation vibration is a common indicator of the health of mechanical systems (engines, gearboxes, rotors, even airframes), for both Helicopters and fixed wing aircraft.

Given the various issues that could afflict a quad (bent rotor blade after impact, high rpm vibration, an unbalanced new rotor, PID misconfiguration etc..) this could be a useful metric for odd warnings or a display element.

I am not sure how complex this could be to implement but could be as simple as infrequently sampling the IMU to limit processing overhead, applying a high pass and a sqrt function.

### Describe alternatives you've considered

Praying my quad is ok after hitting a branch

Quick-Flash, 2025-09-09:
What should we measure though? Each aircraft has a different vibration to others, so how do you decide to normalize what is normal vibration? Should it be raw sensor readings or filtered sensor readings? If its filtered, then its easier to have a level playing ground, if its raw sensor readings its harder to determine how much is too much as many aircraft will behave differently. And if we do filtered sensor readings, what if RPM/Dynamic notch filtering is doing such a good job that we don't have much vibration, but we do have a motor that is going to rip apart due to the prop imbalance? Do we just look at the FFT the dynamic notch calculates and look for anomalies compared to earlier in the flight? If so then a quad that starts out bad wont look good, and the FFT is at the end of the filtering chain.

Its a cool idea, but I just don't know what would be the best way to implement it.

blckmn, 2026-09-27:
We're closing this as part of a clean-up of issues more than a year old. Sorry it didn't get a resolution here, that's on us. If it's still relevant on the latest release, please open a new issue with current details and link back here.

closed 2026-09-27 by blckmn