betaflight/betaflight #14386: VTX PIT Mode not working properly if assigned to same AUX like ARM

notyourbiz, 2025-05-13:
### Describe the bug

So, maybe this is not a bug or i'm simply to stupid - but:

 If i assign VTX PIT Mode to the same switch and AUX like ARM, PIT Mode is not working "out of the Box".
Means: the VTX remains in PIT Mode when the quad gets armed

PIT Mode Aux1 Min950/Max1050 - ARM Aux1 Min1975/Max2050

whenever Arming is prohibited (MSP, Throttle,... whatsoever) switching PIT Mode works fine, except the Quad does not arm.  if arming is allowed, pit mode is not to be disabled, nor the "P" in the OSD disappears.

the only "walkaround", that took me literally ages to figure out,  is to put a delay on the switch (min 0.8s -> arming / min 0.3s -> disarming). 
If i put a shorter delay on arming (~0.3s) Pit Mode deactivates, but the "P" in the OSD (VTX channel - Band:Channel:Pwr:Pit) remains.

you might say: put PIT Mode on a separate channel - problem is, i have none left... ;-)

i can live with my solution for now, but maybe some of you highly respected developers/hackers/code-nerds is interested in that...
and maybe you can put some info about that in the wiki - if already exists, my apologies, i was to dumb to find...

### To Reproduce

Assign VTX Pit Mode to same Aux and Switch like ARM (0-State PitMode / 1-State ARM)

### Expected behavior

working intuitively without tweaking switches in the TX

### Support ID

```txt
4b3cc856-7d1a-4d18-8d4c-8f4d5fc0dfed
```

### Flight controller

 MAMBAF722_I2C

### Other components

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.1 May 12 2025 / 07:20:57 (77d01ba3b) MSP API: 1.46
# config rev: 53e2e0a
# board: manufacturer_id: DIAT, board_name: MAMBAF722_I2C

VTX: TBS Unify Pro32 Nano

### How are the different components wired up (including port information)

correctly, since it works if i put a delay on the TX-switch... ;-)

VTX is on UART3  (TBS SmartAudio) with standard/auto values

### Add any other context about the problem that you think might be relevant here

Had that issue since "ever" - BF 4.2.11 - 4.5.1

notyourbiz, 2025-05-17:
ok folks, here some update from me.

meanwhile i figured out that pit mode is not switchable once the copter is armed. is has nothing to do with all my fancy blabla and experience i wrote. 

when i wrote my 1st post, i was using a s.bus receiver (RM R81), restricted to 8 channels.
now i have an ELRS receiver (RM RP2 v2), which obviously supports more channels and i have an extra switch for pit mode.

in this config now i noticed that switching pit mode, weather on or off,  is not possible once the copter is armed.

don't know if this even is a bug, but maybe for someones interests.

regards

ot0tot, 2025-06-07:
> ### Describe the bug
> So, maybe this is not a bug or i'm simply to stupid - but:
> 
> If i assign VTX PIT Mode to the same switch and AUX like ARM, PIT Mode is not working "out of the Box". Means: the VTX remains in PIT Mode when the quad gets armed
> 
> PIT Mode Aux1 Min950/Max1050 - ARM Aux1 Min1975/Max2050
> 
> whenever Arming is prohibited (MSP, Throttle,... whatsoever) switching PIT Mode works fine, except the Quad does not arm. if arming is allowed, pit mode is not to be disabled, nor the "P" in the OSD disappears.
> 
> the only "walkaround", that took me literally ages to figure out, is to put a delay on the switch (min 0.8s -> arming / min 0.3s -> disarming). If i put a shorter delay on arming (~0.3s) Pit Mode deactivates, but the "P" in the OSD (VTX channel - Band:Channel:Pwr:Pit) remains.
> 
> you might say: put PIT Mode on a separate channel - problem is, i have none left... ;-)
> 
> i can live with my solution for now, but maybe some of you highly respected developers/hackers/code-nerds is interested in that... and maybe you can put some info about that in the wiki - if already exists, my apologies, i was to dumb to find...
> 
> ### To Reproduce
> Assign VTX Pit Mode to same Aux and Switch like ARM (0-State PitMode / 1-State ARM)
> 
> ### Expected behavior
> working intuitively without tweaking switches in the TX
> 
> ### Support ID
> 4b3cc856-7d1a-4d18-8d4c-8f4d5fc0dfed
> ### Flight controller
> MAMBAF722_I2C
> 
> ### Other components
> version: Betaflight / STM32F7X2 (S7X2) 4.5.1 May 12 2025 / 07:20:57 ([77d01ba](https://github.com/betaflight/betaflight/commit/77d01ba3b76a22909d5f09cb0628820141f95eaa)) MSP API: 1.46 config rev: 53e2e0a board: manufacturer_id: DIAT, board_name: MAMBAF722_I2C
> 
> VTX: TBS Unify Pro32 Nano
> 
> ### How are the different components wired up (including port information)
> correctly, since it works if i put a delay on the TX-switch... ;-)
> 
> VTX is on UART3 (TBS SmartAudio) with standard/auto values
> 
> ### Add any other context about the problem that you think might be relevant here
> Had that issue since "ever" - BF 4.2.11 - 4.5.1

Based on your support ID, your FC is also resource starved. F722 FCs should be using a max 4kHz PID loop rate and DSHOT300.

ot0tot, 2025-06-11:
> CPU-Load is a about 62%, Quad is flying darn good... i doubt this is related to my "problem"... ;-)

Show the output of `tasks` in the CLI. You should absolutely not be running an 8kHz PID loop with bidirectional DShot on the F722.

notyourbiz, 2025-06-11:
got your point... the total load is half (exl. serial). but still, the total load @ 8k is absolutely ok imho...
or is there something i've overseen?

not just advice but explanation would be highly appreciated... ;-)

PID loop 8kHz

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       1       0    0.0%    0.0%         0      0     35       1
01 - (         SYSTEM)    997       6       0    0.5%    0.0%         4      3   3479       4
02 - (           GYRO)   8006       6       2    4.8%    1.6%       125      0  28035       0
03 - (         FILTER)   8006      21      16   16.8%   13.5%      1153      0  28035       0
**04 - (            PID)   8006      64      52   51.2%   41.7%      3577      0  28035       0**
05 - (            ACC)    984       8       3    0.7%    0.3%        27      1   3477       6
06 - (       ATTITUDE)    493      13       8    0.6%    0.3%        33      0   1741      11
07 - (             RX)     10      21      17    0.0%    0.0%         4      0    105      18
08 - (         SERIAL)     97    4573       1   44.3%    0.0%       543      0    327      41
09 - (       DISPATCH)    988       6       0    0.5%    0.0%         6      1   3479       3
10 - (BATTERY_VOLTAGE)     50       7       2    0.0%    0.0%         1      0    175       6
11 - (BATTERY_CURRENT)     50       4       0    0.0%    0.0%         0      0    175       3
12 - ( BATTERY_ALERTS)      5       3       2    0.0%    0.0%         0      0     18       3
13 - (         BEEPER)     97       8       3    0.0%    0.0%         2      0    327       7
14 - (           BARO)     40      33      24    0.1%    0.0%        24      0    417      32
15 - (       ALTITUDE)     97       8       2    0.0%    0.0%         1      0    327       6
16 - (      TELEMETRY)    492       6       0    0.2%    0.0%         4      0   1639       4
17 - (            OSD)     12      30      16    0.0%    0.0%        39      4   1617      29
18 - (     ESC_SENSOR)     97       4       1    0.0%    0.0%         0      0    327       2
19 - (            CMS)     20       2       1    0.0%    0.0%         0      0     70       4
20 - (        VTXCTRL)      5       1       1    0.0%    0.0%         0      0     18      31
22 - (    ADCINTERNAL)      1       4       2    0.0%    0.0%         0      0      4       3
23 - (       PINIOBOX)     20       2       1    0.0%    0.0%         0      0     70       2
24 - (SPEED_NEGOTIATION)     97       4       1    0.0%    0.0%         1      0    327       3
RX Check Function                   8       3                         8
Total (excluding SERIAL)                                57.4%

PID loop 4kHz

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (          SYSTEM)     10       5       0    0.0%    0.0%         0      0     56       4
01 - (         SYSTEM)    992       5       0    0.4%    0.0%        12      5   5539       2
02 - (           GYRO)   8000       4       2    3.2%    1.6%       378      0  44426       0
03 - (         FILTER)   4002      22      18    8.8%    7.3%      1809      0  22213       0
**04 - (            PID)   4002      64      47   25.6%   18.9%      4672      0  22213       0**
05 - (            ACC)    999       9       3    0.8%    0.2%        73      1   5538       7
06 - (       ATTITUDE)    499      15       9    0.7%    0.4%        89      4   2771      13
07 - (             RX)     10      24      15    0.0%    0.0%        11      0    165      18
08 - (         SERIAL)    100    5577       2   55.7%    0.0%       560      0    554      44
09 - (       DISPATCH)    996       5       0    0.4%    0.0%        17      1   5540       3
10 - (BATTERY_VOLTAGE)     50       6       2    0.0%    0.0%         2      0    278       4
11 - (BATTERY_CURRENT)     50       5       0    0.0%    0.0%         0      0    278       4
12 - ( BATTERY_ALERTS)      5       5       2    0.0%    0.0%         0      0     28       5
13 - (         BEEPER)    100       8       3    0.0%    0.0%         7      0    554       7
14 - (           BARO)     40      33      22    0.1%    0.0%        68      0    666      31
15 - (       ALTITUDE)    100       9       2    0.0%    0.0%         5      0    554       7
16 - (      TELEMETRY)    499       5       0    0.2%    0.0%        10      6   2769       3
17 - (            OSD)     12      31      18    0.0%    0.0%       105      6   2520      28
18 - (     ESC_SENSOR)    100       6       1    0.0%    0.0%         2      0    554       4
19 - (            CMS)     20       6       2    0.0%    0.0%         0      0    112       5
20 - (        VTXCTRL)      5       1       1    0.0%    0.0%         0      0     27      35
22 - (    ADCINTERNAL)      1       4       3    0.0%    0.0%         0      0      5       6
23 - (       PINIOBOX)     20       5       1    0.0%    0.0%         0      0    112       4
24 - (SPEED_NEGOTIATION)    100       5       1    0.0%    0.0%         3      0    554       3
RX Check Function                   9       3                        27
Total (excluding SERIAL)                                28.4%

closed 2025-07-19 by github-actions[bot]