betaflight/betaflight #14417: Blue leds on top side gone in 4.6.0 (or even 4.5.x)

Nathagamelle, 2025-05-30:
### Describe the bug

The leds are dead if not no 4.4.1 (factory)

### To Reproduce

flash to newer version and see the lack of led after boot

### Expected behavior

blue leds blinking

### Support ID

```txt
bla
```

### Flight controller

2-3S F405 Betafpv

ot0tot, 2025-06-04:
Which LEDs? Which exact FC do you have? Please post a support ID.

Nathagamelle, 2025-06-04:
2-3S F405 BetaFPV https://betafpv.com/collections/brushless-flight-controller/products/f4-2-3s-20a-aio-fc-v1

[image]
What's a support ID?

Nathagamelle, 2025-06-07:
Can confirm in 4.5.2. I have LED STRIP 

[image]

Support ID fe3b28a7-6a5a-4139-8f4d-fd5570d2cb37
When reflashing with Stock, LEDs are back

haslinghuis, 2025-06-07:
In CLI type

```bash
led 0 0,0::CO:10
led 1 1,0::CO:10
save
```

Nathagamelle, 2025-06-07:
That does it. But shouldn't they be added to the base config ?
Likewise, ELRS Rx on UART3 and baro were disabled but could be reenabled in configurator

Thanks