betaflight/betaflight #14427: betaflight 4.5.2 failed ESC reading on 1s Matrix 3in1 HD betafpv

UIsudo, 2025-06-05:
### Describe the bug

esc-configurator.com failed to read ESC when update "1s Matrix 3in1 betafpv" from 4.5.1 to 4.5.2

### To Reproduce

update from stock betaflight 4.5.1 to 4.5.2 choose board: BETAFPVG473
connect 1s battery 
connect to ecs-configurator.com
read setting
-> failed or read very slow

### Expected behavior

reading esc as normal. all 4 esc can be read.

### Support ID

```txt
default
```

### Flight controller

1s Matrix 3in1 HD v1.0 betafpv

mkny, 2025-06-13:
Yes, sure, Even with the battery plugged in. I have to downgrade betaflight to be able to make changes in the bluejay firmware.
Then I upgrade again to the desired betaflight version.
ChatGPT said this ->

> Confirmed: Betaflight 4.5.0/4.5.1 has broken or partial ESC passthrough support on many STM32G4 flight controllers.

UIsudo, 2025-06-13:
[image]

NOt sure what happen, my board works normally when downgrade to 4.5.1. You can change the port USB VCP (see attach picture) to 57600, click save. Remove battery, cable, close betaflight app.  Reconnect battery, cable, go to esc-configurator to check ESC can read again.

haslinghuis, 2025-08-28:
Did anyone check with 4.6 development firmware?

Reports are inconsistent, the hex from BetaFPV uses 4.5.1, so there might be a difference in configuration between the two needed.

cynfewl, 2025-09-22:
I have this same problem with a BetaFPV Air Brushless 5in1 AIO.

When trying to read the ESCs through esc-configurator on 4.5.2 I am unable to access ESC 4 and occasionally ESC 3.

Flashing back to the hex from the BetaFPV website gives me reliable access to all 4 ESCs.

Is there a particular build of 4.6 I should try?

UIsudo, 2025-09-24:
> Build with DSHOT instead of BRUSHED motor protocol

This is 4.5.1 (DSHOT) and ESC reading working
"43db1779-521d-4238-9cdf-e2084093e18b"

This is 4.5.2 (DSHOT) and ESC reading NOT working
"ad85c129-e51b-4825-bba2-802e1686a21a"

This is latest 2025.12.0-beta development build (DSHOT) and ESC reading NOT working
"fa768196-4a1f-4eec-b20f-80c645d1e5d7"

haslinghuis, 2025-09-24:
@UIsudo 

- the 2025-12.0 version is empty, please generate new ID and update.
- with 2025.12.0 also try the core build for testing.

UIsudo, 2025-09-24:
2025.12.0 (DSHOT)
"87ff7b1e-0ca4-4158-943c-6d94428336e9"

2025.12.0 (core only)
"21c7cd80-9b90-43af-9cfd-b023c6532318"

UIsudo, 2025-09-24:
> On 2025.12.0
> 
> ```
> set dshot_bidir = ON
> set motor_pwm_protocol = DSHOT300
> set motor_poles = 12
> save
> ```

updated: 

"2496e496-4aee-471f-ac7b-9c03060c2e63"

UIsudo, 2025-09-24:
> [@UIsudo](https://github.com/UIsudo)
> 
> * Check https://youtu.be/enWZdqV9JQQ
> * See last comment in https://www.reddit.com/r/TinyWhoop/comments/1k7pelf/esc_conf_failed_reading_esc_1/
> 
> <img alt="Image" width="686" height="122" src="https://private-user-images.githubusercontent.com/8344830/493463537-c00ffc43-44c2-4d88-88af-7ab493d149c6.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg3MzQxMzIsIm5iZiI6MTc1ODczMzgzMiwicGF0aCI6Ii84MzQ0ODMwLzQ5MzQ2MzUzNy1jMDBmZmM0My00NGMyLTRkODgtODhhZi03YWI0OTNkMTQ5YzYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDkyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA5MjRUMTcxMDMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Zjc1ODRmODRhMWM3NzM2YzA5ODVjYzNlMGNkMzJhMGFmNzcwN2Q0YjE5NmNmMmJlMjdiZWVmMmJmM2I5ODMwOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.DKjPApjN8TEuHkjSIu-3EuHm7pDmA76owx785mGZEac">

I check the Youtube video and reddit link. I think the problem is with 4.5.2 or 4.6 version or drivers. Many people can read ESCs on 4.5.1 but failed on 4.5.2. Or something changed on Betaflight Passthrough between 4.5.1 and 4.5.2?

cynfewl, 2025-11-25:
Sorry to comment on a closed issue, but this is still very much a problem for me on every firmware version after the BetaFPV supplied  4.5.0.

The BetaFPV versions works fine with esc-configurator, but later versions can't read ESC4 and sometimes ESC3.

Since the only difference is the betaflight version, this seems to be an issue with betaflight rather than the ESC firmware.

cynfewl, 2025-11-26:
It's so cool to see the processes and tools you use on this project! Thank you very much for the prompt response!

I'm looking for #14794 in the configurator app, but it doesn't seem to appear under **Select Pull Request or Commit**. Is there a delay before this shows up in the tool?

EDIT: Never mind, it was user error and I was missing the # on the PR number...

cynfewl, 2025-11-26:
#14794 does appear to fix the issue on the two different boards I have tested.

Reading ESC3 and ESC4 is still a bit slow compared to ESC1 and ESC2 or other flight controllers, but maybe that's something electrical on the BetaFPV boards. I have not had any reads fail across about 30 attempts, mostly in quick succession.

Thank you all!

closed 2025-11-27 by blckmn

zmoothrc, 2026-01-07:
Doesnt work for me either :( reopen please

haslinghuis, 2026-01-08:
@coderabbitai any other suggestions as ESC readings are still an issue

UIsudo, 2026-01-09:
@haslinghuis as the picture above, I already choose DSHOT in Motor Protocol.

ID with dshot_bitbang = ON
Id: "e953a98f-b1f6-4190-a6ef-6705a0dc9f8d"

Also
Could you please consider this information:
"I flashed the firmware provided by betafpv,
[BF4.5.1 G473_M75 Pro O4_CRSF_Matrix HD 3IN1 20250102.hex](https://support.betafpv.com/hc/article_attachments/42317888915097)

this file has version 4.5.1, the ESC reading working correctly.

But, I flashed the same version 4.5.1 from betaflight app then load the setting from betafpv
[BF4.5.1 G473_M75 Pro O4_CRSF_Matrix HD 3IN1 20250102.txt](https://support.betafpv.com/hc/article_attachments/42317919215769)

ESC reading did not work.
"

nerdCopter, 2026-01-09:
> Sorry the fix [#14794](https://github.com/betaflight/betaflight/pull/14794) did not work to me. I flashed the firmware provided by betafpv, [BF4.5.1 G473_M75 Pro O4_CRSF_Matrix HD 3IN1 20250102.hex](https://support.betafpv.com/hc/article_attachments/42317888915097)
> 
> this file has version 4.5.1, the ESC reading working correctly.
> 
> But, I flashed the same version 4.5.1 from betaflight app then load the setting from betafpv [BF4.5.1 G473_M75 Pro O4_CRSF_Matrix HD 3IN1 20250102.txt](https://support.betafpv.com/hc/article_attachments/42317919215769)
> 
> ESC reading did not work.

while relatively old (Nov), noting that vendor provided hex 4.5.1 equals success versus cloud-build 4.5.latest hex equals failure is suspect, IMHO.  wondering if vendor sets a config or resource differently than what is merged into configs repo.

haslinghuis, 2026-04-08:
@Fourflies thanks for your comment. We will investigate.