betaflight/betaflight #14480: Changes in src/main/target/common_pre.h from PR #13814 (#13820) not merged

xs400dohc, 2025-06-26:
### Describe the bug

I also had problems with telemetry on Spektrum SPM4651t (SRXL2) and telemetry. Now I wanted to update to v4.5.2 and had the same problems as before.
After some investigation I found:
PR  #13820 (backport of #13814) not merged correctly. Only changes in src/main/target/common_post.h had been merged, changes in src/main/target/common_pre.h had NOT been merged

### To Reproduce

Flashing v4.5.2 with receiver SRXL2 and telemetry-protocol SRXL still needs user-definition SERIALRX_SPEKTRUM. Otherwise telemetry for Spektrum SPM4651t not working

### Expected behavior

flash with receiver SRXL2 and telemetry SRXL without the need of user-definitions

### Support ID

```txt
54b0693a-47da-4117-a3db-4a4703e66c5e
```

### Flight controller

Speedybee F7V3

osirisinferi, 2025-06-26:
It's not a matter of "not being merged", this was done on purpose in #13947 which was backported to 4.5.x in #13950. See the link to the comment (and the comments below that) in #13947.

xs400dohc, 2025-06-27:
Ah ok, thank you for the information.
I found this, because I flashed to v4.5.2 (rx: SRXL2, tel: SRXL), loaded my dump-file and got these errors:
```
# resource RX_BIND 1 NONE
###ERROR IN resource: INVALID RESOURCE NAME: 'RX_BIND'###

# resource RX_BIND_PLUG 1 NONE
###ERROR IN resource: INVALID RESOURCE NAME: 'RX_BIND_PLUG'###

# set spektrum_sat_bind = 0
###ERROR IN set: INVALID NAME: spektrum_sat_bind = 0###

# set spektrum_sat_bind_autoreset = ON
###ERROR IN set: INVALID NAME: spektrum_sat_bind_autoreset = ON###
```
In the moment I can't proof, if everything is working (my TX is in use by my son). Hopefully, I can test it at this weekend and will report.