betaflight/betaflight #14500: Integrated BMP280 barometer stops working randomly, leading to GPS rescue crashing the drone

lasa01, 2025-07-02:
### Describe the bug

Sometimes when flying with 4.6-dev, I noticed the barometer stops working and the reported altitude stays the same until reboot. Now, this happened during testing GPS rescue, leading to the drone crashing down as the GPS rescue ran the motors on min throttle, trying to descend while the reported altitude doesn't change.

The same FC has worked reliably in another quad on Betaflight 4.4, with no issues with either the barometer or GPS rescue.

### To Reproduce

- Install 4.6-dev commit fd0fd25ef
- Configure and setup baro as well as other things in quad
- See that baro stops working randomly (with my hardware and configuration at least)
- In blackbox logs with baro debug enabled, you can see that all the baro data freezes at random point in flight:

[image]

### Expected behavior

I expect that baro works without problems. In case this happens because of hardware issue with the baro, I expect that Betaflight can detect this and warn me (and maybe fall back to GPS altitude information).

### Support ID

```txt
"301ca777-990f-4d84-aaee-9dbd2383fd61"
```

### Flight controller

Eachine GPS F4 (Tyro119 FC)

### Other components

- GPS: Mateksys m8q-5883
- buzzer
- Ratel 2 camera
- Eachine 4in1 ESC
- Eachine analog VTX
- Happymodel EP1 receiver

### How are the different components wired up (including port information)

Wiring is correct, drone is working fine otherwise. Barometer is built in to FC.

### Add any other context about the problem that you think might be relevant here

I can provide more detailed blackbox logs, etc., if needed.

trenth00, 2025-07-02:
Did the compass freeze too?

trenth00, 2025-07-02:
It might be the same cause as this which is essentially the i2c bus locking up. The PR that this issue mentions greatly reduced the occurrence of the problem but didn’t eliminate it. In my instance I found that the high power telemetry setting seemed to aggravate it more.
https://github.com/betaflight/betaflight/issues/13360

trenth00, 2025-07-02:
Essentially your compass is a bigger culprit in this than your baro. If your Flight controller allows it, put them on their own i2c bus

lasa01, 2025-07-26:
Small update on this, I figured that I can configure another I2C bus, for compass use only, with resource remapping. I checked my flight controller's microcontroller datasheet (STM32F405) and figured out that the motor4 and motor6 pins can be used for I2C_3, using resource command to check existing pin mappings.

I remapped motor 4 to another motor output, and used these commands to clear the timers from the new I2C pins and configure them for I2C:
```
# remap motor 4 to motor 7
resource MOTOR 7 NONE
resource MOTOR 4 B08

# free up motor 6
resource MOTOR 6 NONE

# enable I2C for old motor pins
resource I2C_SCL 3 A08
resource I2C_SDA 3 C09

# clear timers from them, and dma
timer C09 NONE
timer A08 NONE
dma pin C09 NONE
dma pin A08 NONE

# change mag to new I2C
set mag_i2c_device = 3
```

Haven't tested this in flight yet but seems to work fine on the bench!