betaflight/betaflight #14516: HGRLC M100-5883 Magnetometer not detected betaflight 4.6.0-dev

matx93, 2025-07-06:
### Describe the bug

HGRLC M100-5883 Magnetometer is not communicating / not identified in Betaflight 4.6.0-dev on board SPEEDYBEEF405AIO, although correctly wired to SCL/SDA pads. Magnetometer cannot be activated in config tab - setting lost after reboot. GPS works instead ok.

### To Reproduce

Flash 4.6.0-dev, configured ports as below, activated both barometer and magnetometer on config tab. 

### Expected behavior

Should identify Magnetometer when tab is activated. N should show active on dashboard

### Support ID

```txt
741f7f01-4876-4cc5-86e7-9e104d99416f
```

### Flight controller

SPEEDYBEEF405AIO

### Other components

Walksnail Avatar V2
Beeper
ERLS receiver

### How are the different components wired up (including port information)

[image]

matx93, 2025-07-06:
@Pancronos yes I did. Is there a way to check if the custom define was successfully applied?

haslinghuis, 2025-07-06:
@matx93 no did not - can check options in setup tab.

Can add it here:

[image]

matx93, 2025-07-06:
@haslinghuis thanks and sorry. I have not rebuilt it with Magnetometers option.
However issue still applies. 

According to the datasheet, the M100 has the [QMC5883](https://datasheet.lcsc.com/szlcsc/QST-QMC5883L-TR_C192585.pdf)
In [Betaflight's user guide](https://betaflight.com/docs/wiki/guides/current/Magnetometer) the settings are:
- set mag_hardware = QMC5883
- set set mag_i2c_device = 1
- set mag_i2c_address = 13

I have tried all the above and the "N" still is not active in the dashboard

matx93, 2025-07-06:
Tried also with power, still not working.
Support ID: 55d77598-997f-4fb9-81ea-5242464a7204

matx93, 2025-07-15:
@Pancronos Unfortunately still not working. And @haslinghuis is right - also barometer is not working at all. I have tried also disabling the barometer, but still not working

haslinghuis, 2025-07-15:
Does baro work when mag is detached ?

matx93, 2025-07-15:
Nope

haslinghuis, 2025-07-15:
As far I can diagnose the issue this is beginning to look like an (hardware) issue with I2C here.

Logicenios, 2025-07-22:
I also have this GPS Please check the connector wiring on the connector to the pcb where the markings are. Do not use the photos from the site as you might have cut the wrong side of the connector.

matx93, 2025-07-22:
In my case, I have already tried swapping TX/RX or SCA SCL but without success. GPS works, indicating it's communicating with the device

kmastra, 2025-07-22:
I have also tried swapping SCA and SCL. Right now i have those two completely disconnected and GPS is working fine! so i dont think its a wiring issue.

Logicenios, 2025-07-22:
So the problem with i2c starts when you connect the magnetometer. I would try checking the pads that are at the top of the gps with pads that are connected to the FC. Connect everything and check continuity. [image]

surajnkt2001, 2025-07-26:
I also have same issue did you find out any solution

ledvinap, 2025-07-26:
Maybe missing pullups are causing problems. Can you verify there that there is pullup on SCL and SDA?

terryfritz99, 2025-12-05:
> On some boards you also need battery power for the mag to work 0xD (13) is default for QMC5883

I found this out just now after scratching my head for weeks. I have a Speedybee F405v3 and HGLRC M100 and had all sorts of trouble calibrating the mag until I plugged in a battery.