betaflight/betaflight #14523: Safe direct motor control via MSP without motor stop upon disconnection or disarm

zywzywzyw, 2025-07-07:
### Is your feature request related to a problem? Please describe

Hello,

I am conducting research using Betaflight and need to directly control motor speeds via MSP (MultiWii Serial Protocol). Currently, the motors stop if the MSP connection is lost or the system is disarmed, which is not ideal for my research, as I need the motors to continue running even if the MSP connection is interrupted or disarmed. This behavior does not seem safe, especially for response testing.

### Describe the solution you'd like

I would like to know if there is a safer way to directly control motor speeds, ensuring the motors remain running even if MSP disconnects or the system is disarmed.

### Describe alternatives you've considered

Additionally, would switching to Cleanflight offer a better solution for this specific use case?

### Other information

Any advice or suggestions would be greatly appreciated!

Thank you.

zywzywzyw, 2025-07-10:
Thank you for the feedback.

I’ve already tried enabling the failsafe and related protection mechanisms, but unfortunately they did not work in the case of direct motor control via MSP. The motors keep spinning when the MSP connection is lost or the system is disarmed, which is a critical safety issue.

Additionally, I attempted to use a workaround:

I enabled motor_stop so that I could switch to manual control mode (ARM + throttle down) to stop the motors when needed.

However, this also failed in practice during real-world flight tests. It seems that as long as MSP continues sending motor commands, Betaflight does not honor the manual input or mode switching, effectively overriding or ignoring the motor_stop + ARM + throttle low behavior.

So it seems MSP has priority over other inputs, and even failsafe doesn’t kick in when MSP is active. That makes this method unsafe.

Is there any known way to:

Make MSP respect failsafe or system disarm?

Automatically stop motors on MSP disconnection or timeout?

Prevent MSP from overriding user input or motor_stop settings?

Alternatively, does Cleanflight or other firmware offer more granular safety control in this scenario?

Any suggestions or insights would be highly appreciated. Safety is my top concern here.

Thanks again!

haslinghuis, 2025-07-11:
Hint: correct implementation would be to use `USE_RX_MSP_OVERRIDE`

zywzywzyw, 2025-07-12:
> Hint: correct implementation would be to use `USE_RX_MSP_OVERRIDE`

Thanks for the hint. As far as I understand, USE_RX_MSP_OVERRIDE allows overriding channel values (like stick inputs), but not directly overriding motor outputs set via MSP. Is that correct, or am I misunderstanding how it works?

zywzywzyw, 2025-07-12:
When configuring through the https://app.betaflight.com/# web page, I noticed that in the “Motors” tab, enabling motor test mode allows direct control of the motors even while disarmed. This seems to bypass the usual safety logic that prevents motor spin-up when the system is disarmed.

In my case, I control the motors via MSP using the MSP_SET_MOTOR command (ID 214), and I suspect this might follow a similar logic to motor test mode, thus also bypassing the disarm protection. Is this behavior intentional by design?

> 正确。PID 循环和混频器仍然处于活动状态。您可以使用 MSP_SET_MOTOR 命令。无论如何，启动/停止状态都应该始终保留。

amrathod, 2025-07-17:
One strategy would to be keep track of the last time a MSP_SET_MOTOR command was received and in case the update rate drops below the expected rate, set motor_disarmed to zeros. This way if the msp connection breaks, the motors wont keep spinning at the last received command indefinitely. 
It will affect the motor test tab, so there can be a default timeout according to the rate at which the configurator sends the command. I think this would be useful safety feature. Curious to hear the dev team's thoughts.

closed 2025-08-25 by github-actions[bot]