betaflight/betaflight #14535: `master` - make preview fail, maybe others

nerdCopter, 2025-07-12:
### Describe the bug

```
Creating EXST HEX from patched EXST BIN ./obj/betaflight_4.6.0_STM32H730_SPRACINGH7RF.bin, VMA Adjust 0x90100000 
Building target config SPRACINGH7RF succeeded.
cd src/test && /usr/bin/make --no-print-directory test
CC version: Ubuntu clang version 18.1.8 (++20240731025043+3b5b5c1ec4a3-1~exp1~20240731145144.92) Target: x86_64-pc-linux-gnu Thread model: posix InstalledDir: /usr/bin
CXX version: Ubuntu clang version 18.1.8 (++20240731025043+3b5b5c1ec4a3-1~exp1~20240731145144.92) Target: x86_64-pc-linux-gnu Thread model: posix InstalledDir: /usr/bin
compiling ../main/sensors/boardalignment.c 
compiling ../main/common/sensor_alignment.c 
compiling ../main/common/maths.c 
compiling ../main/common/vector.c 
compiling unit/alignsensor_unittest.cc 
In file included from unit/alignsensor_unittest.cc:33:
../../lib/test/gtest/inc/gtest/gtest.h:52:10: fatal error: 'cstddef' file not found
   52 | #include <cstddef>
      |          ^~~~~~~~~
1 error generated.
make[1]: *** [Makefile:814: ../../obj/test/alignsensor_unittest/alignsensor_unittest.o] Error 1
make: *** [Makefile:758: test] Error 2
```

### To Reproduce

.

### Expected behavior

.

### Support ID

```txt
.
```

### Flight controller

.

ledvinap, 2025-07-12:
Seems that libstdc++-14-dev (or other version) is missing on your systex

nerdCopter, 2025-07-12:
> libstdc++-14-dev

confirmed.  i did do moderate cleanup past week. TY

closed 2025-07-12 by nerdCopter