betaflight/betaflight #14545: Change in pre-arm behaviour

schlafsack, 2025-07-15:
### Describe the bug

1. Pre-4.6 betaflight used to display a pre-arm countdown on the OSD but this countdown is not displaying for me in 4.6.

2. If the pre-arm switch is paired with the beacon  (link or range on the same switch) then the countdown is displayed but is stuck - at 0.6 secs in my case.  The drone will not arm.

### To Reproduce

1. Enable warnings in the OSD, configure a pre-arm switch.  No countdown is displayed.

2. Configure pre-arm paired with a beacon on the same switch.  Beacon is displayed as a warning, alternating with the frozen countdown timer - the drone will not arm.

### Expected behavior

1. Countdown should be displayed as a warning in the OSD.

2. Countdown should be displayed as a warning in the OSD. Pre-arm should count down, drone should arm, beacon should sound.

### Support ID

```txt
75324be7-846d-4815-b2a5-818f49656b58
```

### Flight controller

Flywoo GOKU GN405 Nano

etsach, 2025-09-06:
I had the same issue today and can confirm.
Though it is not even required to have a PREARM mode enabled
Any use of the BEEPER mode while arming, prevents arming.

nerdCopter, 2025-09-20:
kind of makes sense to be as is, no?  "beacon" is dshot command.  dshot commands to ESC's to beep seemingly _should_ block arming/

nerdCopter, 2026-02-02:
@coderabbitAI, deep-analyze the code-base about this issue. there were multiple pull-requests that affected dshot since 4.5.x. There were also many other refactoring pull-requests that might have affected this while not being dshot specific.
find root cause and propose solution. fact-check any recommendations.

nerdCopter, 2026-02-02:
@coderabbitAI, but the else statement specifically said `// make sure lastDshotBeaconCommandTimeUs is valid when DSHOT_BEACON_GUARD_DELAY_US elapses` -- is this not true, or is there some valid alternative comment for the code, or instead, some valid alternative code for the comment?