betaflight/betaflight #14573: Position Hold combined with EZ Disarm may result in unintentional disarm

BrandonsBakedBeans, 2025-08-06:
### Describe the bug

I was flying today and I entered position hold mode while moving at a quick pace. Before the drone stabilized I dropped throttle (in order to maintain altitude) and the drone disarmed as it tried to stabilize position hold. I have EZ disarm enabled and I believe that because the throttle was at zero and I was off the cyclic stick AND because it detected a strong enough jolt for ezdisarm to activate it led to disarming mid flight. 
From experience with testing poshold I'm certain that if I hadn't rushed things and kept the throttle mid prior to it jolting to stable hover that it would have stayed armed.

### To Reproduce

Set ezdisarm to 50 and activate position hold while at a decent forward speed, then immediately drop throttle to zero and deadstick roll/pitch before the drone stabilizes.

### Expected behavior

Pos hold should temporarily disable ezdisarm either until its in stable hover or after a set period elapses.

### Support ID

```txt
5987e685-1c9b-430d-8246-bcc758f302bb
```

### Flight controller

Skystars F722 Mini HD2 FC + KM35 ESC

### Other components

BE220 GPS, BetaFPV SuperD ELRS Rx updated to recent firmware, Battery backup Beeper, DJI O4Pro on recent firmware.

tbolin, 2025-08-13:
@haslinghuis I had very littel involvement in the development of easy dissarm, and I'm not sure I can contribute much to this discussion.

ctzsnooze, 2025-09-02:
@BrandonsBakedBeans  This one is tricky.

First up, When the mixer mode is set to EzLanding, there are modifications made to throttle outputs to reduce bouncing around.  This mixer mode is completely unrelated to the 'auto-disarm' function.

The auto-disarm function uses an internal function `disarmOnImpact()`, which is enabled by setting `landing_disarm_threshold` to a non-zero value, and then is checked every PID loop in pid.c, after take-off.

If either:
- RC throttle is below 5%, *and* all sticks are also entered, or
- Altitude Hold mode is active, then
- if the accelerometer has a spike on it's 'acceleration derivative' or 'jerk' signal that exceeds the threshold, it disarms.

Hence while in Altitude Hold mode, the throttle and stick positions are ignored.

If in Altitude Hold mode, the disarm impact sensitivity is increased 50% when below landing altitude, but that increase in sensitivity is not otherwise implemented in normal flight, regardless of altitude.

Note that in normal flight, once the `landing_disarm_threshold` has been set to a non-zero value,  the auto-disarm is possible *at any time in the flight*, if the sticks are in the necessary positions.

It is important that quad should not have a noisy accelerometer signal, or shake badly.  This can happen during descents, or sudden stops, leading to an unwanted automatic disarm.  The accelerometer noise gets added to the impact signal and cannot be distinguished from an impact.  It can be sufficient, alone, with sudden movements that cause the quad to shake, to exceed the threshold.  For these machines, the threshold may need to be set quite high to prevent inappropriate disarms, and then it may not disarm on a real landing.  Generally the auto disarm works best when the accelerometer signal is not very noisy, and the quad doesn't shake on abrupt inputs or stops.

 It's really important that the threshold value is sufficiently high that any normal in-flight movements, no matter how aggressive, and minor collisions with small branches, don't trigger the disarm.  The only way to be sure is to test carefully.  

Generally the higher the threshold, the safer it is while flying, but of course a more solid bump on landing will be needed to disarm.

The Threshold value should therefore be carefully tested over soft grass, where possible triggering events (eg hard flip stops at zero throttle, zero throttle descents, and intentionally striking small branches, are tested, to confirm that they do not cause an unexpected disarm.  

The most likely reason for your disarm, assuming it was via the `disarmOnImpact()` function, was that the hard stop resulted in significant shaking of the quad, exceeding the disarm threshold.  Stick positions would not matter since it was in Altitude Hold mode.  Fundamentally the issue is that your disarm threshold is too sensitive.

Currently we do not track 'how long since Position Hold Mode was initiated', so the sensitivity can't be attenuated on that basis.  

I'm not sure this solves the problem, and indeed I'm not sure there is a solution, but that's the background and the explanation.

Note that the auto-disarm feature is not intended only as an automatic disarm to make landing 'easier' for pilots; it provides an auto-disarm on impact of any kind, to avoid further damage to the quad or say a person or object that might have been hit; it will work even if the radio link fails, eg during failsafe.

I wonder if the CLI name should be changed from `landing_disarm_threshold` to `auto_disarm_threshold` to make this more transparent.

blckmn, 2026-09-27:
Fixed on master by #15701: impact disarm no longer fires while an autopilot mode such as position hold owns the throttle, so a low stick during the hold no longer disarms. It will be in the next release. We're closing this as part of a clean-up of issues more than a year old. If you still see an unintended disarm after that, please open a new issue with a blackbox log and link back here.

closed 2026-09-27 by blckmn