betaflight/betaflight #14590: The recording to the black box on the SpeedyBee F405 V4 is randomly interrupted.

HideM60, 2025-08-14:
### Describe the bug

The combination of BETAFLIGHT 4.6.0 and 4.5.2 with the SpeedyBee F405 V4 frequently causes the BBL to cut off.

[image]

Occasionally, the recording may not resume after being interrupted during flight. 
This phenomenon does not occur with versions 4.5.1 and 4.5.0. There are three aircraft equipped with the SpeedyBee F405V4, and the issue occurs with all of them.

I am Japanese, so this is an automatic translation. I apologize if my English is unclear.

BBL

[LOG00001.zip](https://github.com/user-attachments/files/21764713/LOG00001.zip)

### To Reproduce

Flying with the combination of BETAFLIGHT 4.6.0 or 4.5.2 and the SpeedyBee F405 V4.

### Expected behavior

Normal recording to the black box.

### Support ID

```txt
f0c36584-eb73-4e04-a681-510ac5c32587
```

### Flight controller

SpeedyBee F405 V4

### Other components

SpeedyBee BLS 60A 30x30 4-in-1 ESC
or
SpeedyBee BLS 55A 30x30 4-in-1 ESC

VTX DJI O4

### Add any other context about the problem that you think might be relevant here

I don't know if this is related, but the combination of SpeedyBee F405 V4 and BETAFLIGHT 4.6.0 does not enable "high-capacity storage mode." If you want to retrieve the logs, you need to remove the SD card from the FC every time. 
In 4.5.x, simply enabling "high-capacity storage mode" allows you to extract logs while the SD card is still inserted in the FC.

HideM60, 2025-08-15:
Thank you for the advice.
However, I turned bitbang OFF in the CLI and flew, but it did not improve.

[image]

BBL

[LOG00002.zip](https://github.com/user-attachments/files/21791366/LOG00002.zip)

HideM60, 2025-08-15:
ee74bb21-195c-485e-b4df-e6009e5adf73

HideM60, 2025-08-15:
> > アドバイスありがとうございます。ただ、CLIでbitbangをOFFにして飛ばしてみましたが改善しませんでした。
> > [image]
> > BBL
> > [LOG00002.zip](https://github.com/user-attachments/files/21791366/LOG00002.zip)
> 
> 新しいサポート ID を投稿できますか?

ee74bb21-195c-485e-b4df-e6009e5adf73

loghatVector, 2026-03-10:
It seems like the issue you're experiencing with the black box recording interruptions may be related to changes in how Betaflight 4.6.0 and 4.5.2 handle storage and logging. First, I recommend checking the SD card for any corruption or compatibility issues, as a faulty card can cause recording interruptions. You can also try formatting the SD card using FAT32 and testing it again.

Additionally, since you mentioned that "high-capacity storage mode" is not enabled, ensure that the SD card meets the required specifications for Betaflight. For further diagnostics, analyze the provided log file using LogHat (loghat.app) to identify any anomalies or errors during the recording sessions. This could provide insight into whether the issue is related to the firmware or hardware.

blckmn, 2026-09-27:
#15682, now on master, fixes a long-standing DMA bug. Clearing one stream's flags wiped the flags of other streams on the same controller, and left its own error flags set. The SD card on this board runs over SPI with DMA, which that bug affects, so it may be related. The bug was also present in 4.5.1, though, so it doesn't explain why 4.5.1 was fine, and we're not certain.

Would any of you be prepared to flash a development build and tell us whether the blackbox still cuts off? In the app's Firmware Flasher, tick "Show release candidates" and choose the "Development" build type. A log from a flight where it stops, plus a support ID, would help either way.

brobberfpv, 2026-10-20:
I tested an **unmodified master development build that includes #15682**, but Blackbox logging still stopped after some time. `dshot_bitbang = OFF` did not resolve it.

I don’t have hardware expertise, so I used **Codex** to help investigate. The experimental recovery code described below was suggested by Codex.

The SD card’s SPI mode changed from **mode 0 to mode 3** in #14142, backported to 4.5.2 in #14143. This matches the reported version boundary: 4.5.1 works, while 4.5.2 has the problem.

I compared both modes on my **SpeedyBee F405 V4** using diagnostic firmware. The only functional difference between the comparison builds was the SD SPI mode.

**All tests were performed on the bench with the props removed.** Each mode-comparison test lasted approximately four minutes:

- **Mode 0:** clean log, no SD errors or missing samples.
- **Mode 3:** 210 invalid SD write responses (`0xCA`) and 10 gaps in the log, including one lasting about two seconds.
- **Back to mode 0:** clean again, no SD errors or missing samples.

Both diagnostic builds included the same TX-only DMA/chip-select changes and experimental recovery code. The recovery code retries a failed runtime CMD8 check within the existing retry limit instead of immediately disabling the SD card. In the mode-3 test, writing resumed after a captured CMD8 failure, but gaps remained in the log. The mode-0 tests recorded no SD errors and did not need recovery.

This strongly suggests that the SPI mode change contributes to the problem on my setup. It is not yet a confirmed fix for every board/card combination.

I have preserved the Blackbox logs and CLI diagnostics for all three comparison tests. If time permits, I plan to compare stock 4.5.1 with a version of 4.5.1 modified only to use SD SPI mode 3.

Hope it helps.

brobberfpv, 2026-10-22:
I tested a build based on official Betaflight 2026.6.2 on my SpeedyBee F405 V4.

I added a persistent CLI setting, sdcard_spi_mode, to compare modes 3 and 0 using the same firmware. This build keeps the stock SD transfer and error handling, without the experimental recovery or DMA changes used in my earlier diagnostic builds.

I tested mode 3 first, then mode 0:
- Mode 3 — LOG00002: 40 gaps and 3,265 missing or undecodable samples within 93.9 seconds of recorded data. No normal end marker.
- Mode 0 — LOG00003: All 480,394 expected samples over 241.6 seconds. No gaps or corrupt frames, and a normal ending.

I propose adding sdcard_spi_mode as a CLI setting:
- Default 3, preserving current behavior.
- Accept 0 and 3 only.
- 0 to cover case like this, it is anyways used in the past, should be pretty much safe IMO

On the modified firmware, mode 0 can be selected with: 

set sdcard_spi_mode = 0

This option could help in cases like mine, where logging stops for an unknown reason. Betaflight already used mode 0 in the past, so making it available again as an optional setting seems reasonable, while keeping mode 3 as the default.

If needed I can attach BB logs.