betaflight/betaflight #14592: Suggested algebraic simplicication of Dynamic Idle PID DTerm

martinbudden, 2025-08-14:
### Is your feature request related to a problem? Please describe

Not a feature request, just a suggestion for a code simplification.

In `calculateThrottleAndCurrentMotorEndpoints` we have

```c
// PT1 type lowpass delay and smoothing for D
minRps = mixerRuntime.prevMinRps + mixerRuntime.minRpsDelayK * (minRps - mixerRuntime.prevMinRps);
float dynIdleD = (mixerRuntime.prevMinRps - minRps) * mixerRuntime.dynIdleDGain;
```

substituting the first line into the second line gives:

```c
float dynIdleD = (mixerRuntime.prevMinRps - (mixerRuntime.prevMinRps + mixerRuntime.minRpsDelayK * (minRps - mixerRuntime.prevMinRps))) * mixerRuntime.dynIdleDGain;
```

ie

```c
float dynIdleD = -mixerRuntime.minRpsDelayK * (minRps - mixerRuntime.prevMinRps) * mixerRuntime.dynIdleDGain;
```

ie

```c
float dynIdleD = -mixerRuntime.dynIdleDGain*mixerRuntime.minRpsDelayK * (minRps - mixerRuntime.prevMinRps);
```

Now, since `dynIdleDGain` and `minRpsDelayK` are always used together, in `mixerInitProfile` we can replace

```c
    mixerRuntime.dynIdleDGain = currentPidProfile->dyn_idle_d_gain * 0.0000003f * pidGetPidFrequency();
...
    mixerRuntime.minRpsDelayK = 800 * pidGetDT() / 20.0f; //approx 20ms D delay, arbitrarily suits many motors
```

with

```c
    mixerRuntime.dynIdleDGain = currentPidProfile->dyn_idle_d_gain * 0.0000003f * pidGetPidFrequency() * 800 * pidGetDT() / 20.0f;
```

and get rid of `mixerRuntime.minRpsDelayK` altogether.

### Describe the solution you'd like

As in comment above.

### Describe alternatives you've considered

Do nothing.

ledvinap, 2025-08-16:
@martinbudden : Problem is that you still need to calculate filtered minRps (to set prevMinRps). Without it you remove PT1 part (and apply wrong scaling to D term)

martinbudden, 2025-08-17:
@ledvinap , yes it did seem a bit too good to be true. Thanks for pointing out the oversight.

Closing the issue.

closed 2025-08-17 by martinbudden