betaflight/betaflight #14636: v4.6 iFlight Blitz F722 arm but no throttle control

ExFPVengineer, 2025-08-31:
### Describe the bug

I can confirm similar behavior on Betaflight 4.6 dev with Crossfire.

FC: iFlight Blitz F722

Receiver: TBS Crossfire

ESC protocol: DShot600, bidirectional enabled

Configurator: 10.10.x

Issue: Quad arms normally, receiver tab shows correct Crossfire stick ranges (1000–2000), and motors spin correctly in the Motors tab. However, after arming, throttle has no effect on the motors. Roll/pitch/yaw channels are recognized, but throttle input is ignored.

Notes:

Rolling back to 4.5.2 stable resolves the problem immediately with the same config/dump.

Behavior is reproducible across multiple flashes.

Tested both with and without bidirectional DShot enabled → no change.

### To Reproduce

Steps to reproduce the issue:

Flash FC to Betaflight 4.6 dev.

Apply working configuration (dump attached in the support ID)

Arm the quad.

Raise throttle stick.

Observe no change in motor output.

### Expected behavior

Throttle input to reflect motor output.

### Support ID

```txt
# # Betaflight / STM32F7X2 (S7X2) 4.5.2 Mar 20 2025 / 17:13:35 (024f8e13d) MSP API: 1.46

# # config rev: 0f78778

# # start the command batch

# batch start
Command batch started

# # reset configuration to default settings

# defaults nosave

# resetting to defaults

# board_name IFLIGHT_BLITZ_F722
board_name IFLIGHT_BLITZ_F722

# manufacturer_id IFRC
manufacturer_id IFRC

# mcu_id 001a0007363250172037314e
mcu_id 001a0007363250172037314e

# signature
signature

[... 416 lines cut ...]
roll_srate set to 50
# set pitch_srate = 50
pitch_srate set to 50
# set yaw_srate = 50
yaw_srate set to 50
# rateprofile 1
rateprofile 1

# rateprofile 2
rateprofile 2

# rateprofile 3
rateprofile 3

# # restore original rateprofile selection

# rateprofile 0
rateprofile 0

# # save configuration
```

### Flight controller

iFlight Blitz F722

haslinghuis, 2025-09-04:
All - provide a support ID like requested when opening this issue.

jcink, 2025-09-06:
Thank you, I was able to get throttle response again and all is well.

sjauquet, 2025-09-08:
Had the same problem, but I have made many tests before finding this bug report and here are my findings:

Throttle blocked when arming directly in rateprofile 0 (Betaflight 4.6.0-beta)
Flight Controller: SpeedyBee F405 Mini (SPEEDYBEEF405MINI)
Firmware: Betaflight 4.6.0-beta (commit 78e95edf9, build 2025.12.0-beta)
Configurator: Betaflight Configurator 10.10+
Protocol: SBUS
Features enabled: GPS, OSD, Airmode, Anti-gravity

Description of the bug

When arming directly in rateprofile 0, motors spin at idle but throttle stick input is ignored → the quad will not take off.

If I switch to rateprofile 1, 2 or 3, throttle works normally.

If I arm in rateprofile 1 and then switch back to rateprofile 0 while armed, throttle works in profile 0.

If I clone profile 1 into profile 0, the issue disappears (until modified with the Graphical interface).

This suggests a bug in the initialization of rateprofile 0 at boot/arming.

Steps to reproduce

Flash Betaflight 4.6.0-beta (commit 78e95edf9) on SpeedyBee F405 Mini.

Set default rateprofile to 0.

set rateprofile = 0
save

Arm the quad while in rateprofile 0.

Observed: Motors spin at idle, throttle input ignored.

Expected: Throttle should respond normally.

Switch to rateprofile 1 → throttle works.

Switch back to rateprofile 0 → throttle works again.

Observations

throttle_limit_type = OFF and throttle_limit_percent = 100 in all profiles.

thr_mid, thr_expo, thr_hover identical across profiles.

AUX modes (ARM, ANGLE, GPS RESCUE, FLIP AFTER CRASH, etc.) do not differ between profiles.

Only arming directly in profile 0 triggers the bug.

Likely an initialization issue with profile 0 at boot.

rateprofiles:

profile 0

rateprofile 0

# rateprofile 0
set rateprofile_name = 140-530
set roll_rc_rate = 14
set pitch_rc_rate = 14
set yaw_rc_rate = 16
set roll_expo = 55
set pitch_expo = 55
set yaw_expo = 55
set roll_srate = 53
set pitch_srate = 53
set yaw_srate = 43

rateprofile 1

# rateprofile 1
set rateprofile_name = 170-630
set roll_rc_rate = 17
set pitch_rc_rate = 17
set yaw_rc_rate = 20
set roll_expo = 55
set pitch_expo = 55
set yaw_expo = 55
set roll_srate = 63
set pitch_srate = 63
set yaw_srate = 53

rateprofile 2

# rateprofile 2
set rateprofile_name = 170-800
set roll_rc_rate = 17
set pitch_rc_rate = 17
set yaw_rc_rate = 20
set roll_expo = 55
set pitch_expo = 55
set yaw_expo = 55
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 70

rateprofile 3

# rateprofile 3
set rateprofile_name = 10-310E
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 31
set pitch_srate = 31
set yaw_srate = 36

# restore original rateprofile selection
rateprofile 0

mituritsyn, 2025-09-09:
> Just tested master. No issue with arming or throttling up.
> 
I assume the issue comes from rc link preset
this is the values after applying elrs 150Hz HD frestyle

set rc_smoothing = ON 
set rc_smoothing_auto_factor = 175 
set rc_smoothing_auto_factor_throttle = 100 
set rc_smoothing_setpoint_cutoff = 12 
set rc_smoothing_throttle_cutoff = 20

closed 2025-10-17 by github-actions[bot]